adafruit/circuitpython #9690: ReadtheDocs: displayio.Display class documentation mission from 9.1.x versions

wminarik, 2024-10-06:
### CircuitPython version

```python
documentation
```

### Code/REPL

```python
none
```

### Behavior

The description of the displayio.Display class is missing from the 9.1.x branch ReadtheDocs pages, but is present in earlier versions: e.g. https://docs.circuitpython.org/en/8.2.x/shared-bindings/displayio/index.html
Perhaps it was lost during the reorganization of displayio for version 9.

dhalbert, 2024-10-06:
`displayio.Display`, `.EPaperDisplay`, `.FourWire`, and `.I2CDisplay` are deprecated in 9.x.x and will be removed in 10.0.0. That is why they were removed from the documentation. Perhaps `displayio` should contain cross-references to the new top-level modules: `epaperdisplay`, `fourwire`, `i2cdisplaybus`, and `paralleldisplaybus`, and `busdisplay`.

wminarik, 2024-10-06:
Thanks Dan -and on a weekend too...
I think cross-references are essential -there is a tremendous amount of documentation that references the depreciated way of doing things. Knowing now to look for the `busdisplay` docs helps to put the pieces together. Much of this complexity is hidden on boards with integral displays, but when adding displays to other boards, it's essential.
Old syntax:
```
display_bus = displayio.FourWire(spi, command=tft_dc, chip_select=tft_cs, reset=tft_rst)
display = gc9a01.GC9A01(display_bus, width=240, height=240, rowstart=1, colstart=2,
                        backlight_pin=tft_bl, auto_refresh=False)
```
New syntax (I think):
```
display_bus = fourwire.FourWire(spi, command=tft_dc, chip_select=tft_cs, reset=tft_rst)
display = gc9a01.GC9A01(display_bus, width=240, height=240, rowstart=1, colstart=2,
                        backlight_pin=tft_bl, auto_refresh=False)
```
but the only place where the options in the display constructor (`width`, `height`, `auto_refresh`, etc.) are documented currently is under`busdisplay` -and no way to know to look there. These options are not mentioned or referenced on the `fourwire` page.

That is, the documentation needs to link the old way to the new way to avoid much confusion, at least for me!

dhalbert, 2025-03-22:
Fixed by #10160.

PR #10160 (2025-03-21): Fix multiple documentation issues.
- Fixes #10037.
- Fixes #10147.
- Fixes #9690.
- Fixes #9733.
- Fixes #7943.
- Fixes #7743.
- Fixes #4493.

Also fixes a few minor typos I encountered elsewhere.
closed 2025-03-22 by dhalbert