adafruit/circuitpython #9691: exit_and_deep_sleep_until_alarms() crashes on ESP32-S3 Feather

skjdghsdjgsdj, 2024-10-06:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.4 on 2024-09-17; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3
```

### Code/REPL

```python
# full context: https://github.com/skjdghsdjgsdj/babypod-software/blob/main/power_control.py#L65
alarm.exit_and_deep_sleep_until_alarms(pin_alarm, time_alarm, preserve_dios = [i2c_power])
```

### Behavior

The board resets into safe mode and requires a reset. The console output shows:

```
Enabled interrupt for center button
Cleared encoder read queue
Waiting a few seconds for deep sleep
Creating time alarm
Preserved I2C_POWER
Created pin alarm
Entering deep sleep; will wake up from PinAlarm or TimeAlarm in 59 seconds

[12:40:50.980] Disconnected
[12:40:52.003] Connected to /dev/tty.usbmodem[...]
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

I'm attempting to enter deep sleep on an ESP32-S3 Feather. It always crashes to safe mode, even when I'm connected by USB. That is, it doesn't try to enter the fake deep sleep when USB is attached, or it doesn't think USB is attached even though I'm in a USB serial console with macOS's `tio`.

Here is how I'm using deep sleep: https://github.com/skjdghsdjgsdj/babypod-software/blob/main/power_control.py#L65

I've tried the following:

* Not passing any `preserve_dios` pins
* Only a TimeAlarm
* Only a PinAlarm

This is how my board is wired: https://github.com/skjdghsdjgsdj/babypod-hardware/blob/main/docs/img/fritzing.png

The idea is an interrupt on the rotary encoder, wired to the INT pin on the rotary encoder board to the Feather's D11 pin, will exit deep sleep and therefore act as a power button. I preserve I2C_POWER because otherwise power is lost to the rotary encoder and it can't raise an interrupt.

I've been resetting the board by entering the REPL and using:
```
import microcontroller
microcontroller.reset()
```
It's unclear to me if that method of reset differs in any meaningful way from pressing the physical reset button.
 My Feather is mounted in an enclosure and the physical reset button on the Feather isn't accessible.

I also programmatically cut power to the STEMMA QT bus via the following and reset again, but it made no difference, hoping it would reset the rotary encoder board if that was being the problem:

```
>>> import digitalio
>>> import board
>>> i2c_power = digitalio.DigitalInOut(board.I2C_POWER)
>>> i2c_power.switch_to_output(True)
>>> i2c_power.value = False # this cut power given my LCD turned off immediately
>>> import microcontroller
>>> microcontroller.reset()
```

My Feather is connected to a battery so it's always supplied power, whether or not connected to USB.

dhalbert, 2024-10-06:
Could you retest with the absolute latest build, which is past 9.2.0-beta.0?
https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/adafruit_feather_esp32s3_4mbflash_2mbpsram/

skjdghsdjgsdj, 2024-10-06:
Same problem and same output with the latest snapshot:

```
dafruit CircuitPython 9.2.0-beta.0-25-g7f14b9c49a on 2024-10-04; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3
```

skjdghsdjgsdj, 2024-10-07:
I believe this is related to: https://github.com/adafruit/circuitpython/issues/7447

I have a global exception handler in my `code.py`. This commit makes deep sleep perform as expected when connected to USB: https://github.com/skjdghsdjgsdj/babypod-software/commit/061031acb539283232860684e88eaaf88ca12ee4

Output is now:

```
Enabled interrupt for center button
Cleared encoder read queue
Waiting a few seconds for deep sleep
Creating time alarm
Preserved I2C_POWER
Created pin alarm
Entering deep sleep; will wake up from PinAlarm or TimeAlarm in 59 seconds

Code done running.
Auto-reload is off.

Press any key to enter the REPL. Use CTRL-D to reload.
Pretending to deep sleep until alarm, CTRL-C or file write.
Woken up by alarm.
```

...and pressing the button on the rotary encoder to trigger the interrupt wakes from deep sleep.

I did try catching `BaseException` and just immediately `pass`, but the mere act of catching it in the first place seems to cause this bug.

dhalbert, 2024-10-07:
As mentioned in #7447, internally, requesting deep sleep raises `DeepSleepRequest`, which is a subclass of `BaseException`. It should not crash, but as you found, catching `Exception` instead is a work-around.

PR #9692 (2024-10-07): Avoid crashing when printing DeepSleepRequest objects
Before, this would crash trying to print a NULL object. `mp_raise_type` is the correct way to raise an exception without an associated object or message.
```
>>> import alarm
>>> try:
...     alarm.exit_and_deep_sleep_until_alarms()
... except BaseException as e:
...     ee = e
... 
>>> ee
DeepSleepRequest()
```

I also looked at including the deep sleep arguments in the args, but it was confusing because it would seem to print the TimeAlarm rather than the DeepSleepRequest:
```
=== import alarm
=== try:
===     alarm.exit_and_deep_sleep_un [... cut]
closed 2024-10-07 by dhalbert