adafruit/circuitpython #9693: time.sleep waits less than expected

ishahak, 2024-10-07:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0-beta.1-48-g98a1083d6-dirty on 2023-08-22; ESP32-S3-DevKitC-1-N32R8 with ESP32S3
```

### Code/REPL

```python
import time

LOOP=200
acc = 0
for i in range(LOOP):
    t = time.monotonic_ns()
    time.sleep(0.01)
    t2 = time.monotonic_ns()
    diff_us = (t2-t) // 1000
    acc += diff_us
avg = acc / LOOP
print(f'expected 10000 us, got {avg} us')
```

### Behavior

The result shows:
expected 10000 us, got 9799.07 us

### Description

see:
https://forums.adafruit.com/viewtopic.php?p=1024856#p1024856

todbot, 2024-10-07:
You should try that on CircuitPython 9.1 or better.
There have been many updates to the ESP32 internals since 8.1.

anecdata, 2024-10-07:
Similar in 9.1.4. A sliver of the loss (<8us) is from the time it takes to call `time.monotonic_ns()`. There also seems to be a tinier sliver lost to the float division. Without any division::
```
expected = 2_000_000_000
actual      = 1_961_395_269 (avg 9806.98 us)
```

ishahak, 2024-10-07:
Re @todbot, if you followed the linked forum discussion you'll see that bug was reproduced on 9.1.1 as well

todbot, 2024-10-07:
Right, thanks I see that now. Also, thanks @anecdata.  In my experience trying to do tight-timing on CircuitPython this isn't very surprising. CircuitPython has no concept of atomic or critical sections so who knows what happens during those three function calls: USB handling and garbage collection come to mind as tasks that eat up several milliseconds.  

Perhaps a similar test that was disconnected from USB, does a `gc.disable()` before starting, and then logs the final results to a file would give you a better indication as to true timing.  That's not very satisfactory, but I've not found a way for CircuitPython to not be 5-10 milliseconds sloppy with timing. That you're only getting 0.2 milliseconds variation is pretty great, to me.

dhalbert, 2024-10-07:
What I think is odd here is that the time is _less_ that the requested time. I'd expect overhead to make it longer. So I'm suspecting a ticks-vs-msec issue: a tick is 1/1024 of a second vs the msec 1/1000 of a second.

todbot, 2024-10-07:
Oh yes, under the requested value. You are correct. I need more coffee.

johnnohj, 2024-10-10:
I think adafruit_support_bill got it right way back in August: [here](https://forums.adafruit.com/viewtopic.php?p=1025014#p1025014).  Plus, any short sleep will be accentuated by the floored division `diff_us = (t2-t) // 1000` which rounds decimal remainders down to the nearest integer, e.g., 1.999999 -> 1 (per the [docs](https://docs.python.org/3/library/stdtypes.html#numeric-types-int-float-complex)).  See also [this](https://docs.python.org/3/tutorial/floatingpoint.html#floating-point-arithmetic-issues-and-limitations) for a detailed discussion of what adafruit_support_bill and @dhalbert describe. (Of interest might be the point made about perhaps seeing greater accuracy, in this case, with `time.sleep(0.1)` instead)

todbot, 2024-10-11:
> I think adafruit_support_bill got it right way back in August: [here](https://forums.adafruit.com/viewtopic.php?p=1025014#p1025014). Plus, any short sleep will be accentuated by the floored division [...]

Ah yes, I have experienced this. In #9237 I was convinced it was a problem with floats in CircuitPython because 0.004 seconds / 1000 != 4 milliseconds.

johnnohj, 2024-10-11:
> Ah yes, I have experienced this. In #9237 I was convinced it was a problem with floats in CircuitPython because 0.004 seconds / 1000 != 4 milliseconds.
> 

My favorite quote from the discussion:

>"Note that this is in the very nature of binary floating point: this is not a bug in Python, and **it is not a bug in your code either.** You’ll see the same kind of thing in all languages that support your hardware’s floating-point arithmetic" (emphasis mine) 
>

dhalbert, 2024-10-11:
> Plus, any short sleep will be accentuated by the floored division `diff_us = (t2-t) // 1000` which rounds decimal remainders down to the nearest integer, e.g., 1.999999 -> 1 (per the [docs](https://docs.python.org/3/library/stdtypes.html#numeric-types-int-float-complex)).

I think this is still something to follow up on. We might round instead of truncating. And I am still suspicious of 1024 vs 1000. 9.7 vs 10 is not a few least-significant bits of difference.

johnnohj, 2024-10-11:
> I think this is still something to follow up on. We might round instead of truncating. And I am still suspicious of 1024 vs 1000. 9.7 vs 10 is not a few least-significant bits of difference.
>

I had played around with altering the value given time.sleep; the handful of substitute values tried were enough to suggest that expressions are perhaps being evaluated according to _The Price is Right_ rules, which is to say what it returns is as near as possible the specified value without going over, within the limits of floating point/ (1000/1024) math. 

Using values for sleep from 0.08 to 0.011 returned results which were either 986x(- up to 90) or 1076x(- up to 20). 

Using 0.1 (and adjusting divisor to 10000) returned 9985(- up to 20).

(These tests were done on a Metro RP2040 running 9.2.0-beta.0)

eightycc, 2025-03-28:
`mp_hal_delay_ms()` starts out by calculating `end_tick` by converting `port_get_raw_ticks()` to ms, but adjusts remaining ticks using `port_get_raw_ticks()` without conversion.

johnnohj, 2025-03-28:
> but adjusts remaining ticks using `port_get_raw_ticks()` without conversion.

Does this mean it's literally losing track of time? 

(fwiw: Revisiting this thread led me to imagine potential build flags for fuzzy logic. I'm too new to the project to know, but I have to think it's been discussed somewhere)

Neradoc, 2025-03-29:
> `mp_hal_delay_ms()` starts out by calculating `end_tick` by converting `port_get_raw_ticks()` to ms, but adjusts remaining ticks using `port_get_raw_ticks()` without conversion.

I only see the delay being converted from ms to ticks. The rest is all done in ticks.
https://github.com/adafruit/circuitpython/blob/25216c88c7083fba1162946e4a5c1271f4bf5714/supervisor/shared/tick.c#L102

The issue *is not with floats*, it's because the actual sleep time is in milliseconds, and there's 1024 ticks in 1000 milliseconds, so it just can't be accurate.
- 0.01 gets converted to 10 milliseconds *correctly*
- `mp_hal_delay_ms()` gets called with 10ms as input *correctly*
- 10ms gets converted to 10 ticks because in int land: `(10 * 1024) / 1000 = 10` which is a loss of accuracy
- the board _correctly_ sleeps for 10 ticks, which are actually ~9.77ms (10 * 1000 / 1024)

The only way to have a higher precision would be to use subticks (which are a 1/32th of a tick or ~31us).
10 ms = 10 * 1024 / 1000 = 10.24 ticks = 10 ticks + 8 subticks.

eightycc, 2025-04-02:
> I only see the delay being converted from ms to ticks. The rest is all done in ticks.

@Neradoc Right. I'd mis-read the code. It does boil down to a question of precision as you describe.

eightycc, 2025-04-03:
Using subticks to increase timer resolution by 5 bits, results running on a Raspberry Pi Pico W are:
```
10.0.0-alpha.1-19-g380b8bda7a-dirty
expected 10000 us, got 9997.95 us
```
I'll be making similar updates to other ports and will PR shortly.

dhalbert, 2025-04-03:
> For CP, do we guarantee it's at least the time given (by rounding up) ? Do we say it's a "closest" effort plus overhead ? Do we just say it's an approximation ? Do we promise nothing ?

No guarantees as to whether it's a ceiling rather than a floor, because there are things that can make the interval longer, like gc and interrupt handling. Right now `time.sleep(0.00001)` doesn't really do anything. For more precise timing, one can use `microntroller.delay_us()`, which does a busy-wait loop. Perhaps @eightycc's changes will subsume that.

eightycc, 2025-04-03:
> I've seen library code with time.sleep(0.00001) or 0.0001 which I don't know if this results in an actual sleep of 1 tick or 0.

The conversion code in `time_sleep` first converts the floating time in seconds to milliseconds by floating multiplication and rounds up by adding 0.5. An implicit cast from float to int then truncates the fractional part (standard C). So, a value < 0.5 milliseconds will not sleep while a value >= 0.5 milliseconds will sleep at least 1 millisecond.

> For more precise timing, one can use microntroller.delay_us(), which does a busy-wait loop. Perhaps @eightycc's changes will subsume that.

My update does not support sleep timer resolution beyond integral milliseconds.

Since the underlying timer has subtick resolution (in most cases), it is possible to resolve sleep to ~30 us. Perhaps change `mp_hal_delay_ms` to `mp_hal_delay_us`? This would avoid any busy waiting. Is there utility in doing this?

dhalbert, 2025-04-03:
> My update does not support sleep timer resolution beyond integral milliseconds.
>
> Since the underlying timer has subtick resolution (in most cases), it is possible to resolve sleep to ~30 us. Perhaps [...]

Are you saying change that call? `mp_hal_delay_us()` typically is a busy-wait, not a tick wait (which will save power, since it uses WFI() or the equivalent.

One issue is if a very long sleep would overflow the 32-bit arg to `mp_hal_delay_us()`.
There could be a call in the `time.sleep()` impl to call mp_hal_delay_ms()` for the ms part and then call mp_hal_delay_us()` for the rest.

eightycc, 2025-04-03:
> Are you saying change that call? mp_hal_delay_us() typically is a busy-wait, not a tick wait (which will save power, since it uses WFI() or the equivalent.

Because there is not a generic `mp_hal_delay_us()`, I assumed it didn't exist. My bad. I agree that a busy wait is not a good idea due to power consumption implications. I'll choose another name that clarifies whether the implemented delay is a busy or a timer wait.

tannewt, 2025-05-01:
@eightycc Where did you end up on this?

eightycc, 2025-05-01:
@tannewt It got pushed out by other stuff. I'd meant to make a more comprehensive fix, but that got complicated fast. I do have a smaller fix that addresses the original issue without increasing resolution to delays less than 0.01 second.

closed 2025-05-08 by tannewt