adafruit/circuitpython #9695: rotaryio on RP2350 Non-functional

dcooperdalrymple, 2024-10-08:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.0-beta.0 on 2024-09-17; Raspberry Pi Pico 2 with rp2350a
```

### Code/REPL

```python
import rotaryio
import time
import board

enc = rotaryio.IncrementalEncoder(board.GP16, board.GP17)
last_position = None
while True:
    position = enc.position
    if last_position == None or position != last_position:
        print(position)
    last_position = position
```

### Behavior

Print initial `0` and then no changes to `enc.position` after interacting with rotary encoder.

### Description

- This error seems to only be affecting RP2350 (likely related to PIO)
- Works on "Adafruit CircuitPython 9.2.0-alpha.2351 on 2024-08-22; Raspberry Pi Pico 2 with rp2350a" (also tested on Pimoroni Pico Plus 2 with rp2350b)
- RP2040 (Raspberry Pi Pico) does not exhibit this problem on both 9.2.0-alpha.2350 or 9.2.0-beta.0

### Additional information

I initially thought this was a hardware issue, but scoping it out reveals no problems on 9.2.0-alpha or 9.2.0-beta. I'm not sure of the actual source of this issue, but I may do some additional investigation on my own in the near future.

dhalbert, 2024-10-08:
This was fixed already: see #9681 and the fix #9682. This will be in the next beta, or you can get an "Absolute Newest" build.

closed 2024-10-08 by dhalbert

dcooperdalrymple, 2024-10-08:
Ah, my bad. Thank you for pointing this out!