adafruit/circuitpython #9697: USB host dual role

p3lim, 2024-10-08:
Hi,

I'd like to see support for USB host dual role in CircuitPython, something that was mentioned on the Feather product page is already supported in Arduino. Arduino seems to be using TinyUSB for this, the same as CircuitPython.

My specific use-case is this; I use CircuitPython to drive a keyboard, and I'd like to extend this keyboard with one or two USB host ports. I'd connect a mouse to one port, and a more generic HID device to the 2nd port (like a USB flash drive or a yubikey).

Some of the work seems to be done already in #5986, I was able to listen to a mouse and yubikey OTP (which just simulates key presses) just fine on a RP2040-based board with a USB port (like the Feather) using usb.core, so "echoing" the mouse input through to the computer should be possible, but the passthrough isn't here yet (e.g. for yubikey HID CDC, FIDO2/U2F).

samblenny, 2024-10-08:
Could you give more detail about what you tried, what worked, and what didn't work? Maybe with some sample code?

On the side of CircuitPython acting as USB host, were you trying to use to use [usb.core.find()](https://docs.circuitpython.org/en/latest/shared-bindings/usb/core/index.html#usb.core.find), [usb.core.Device.set_configuration()](https://docs.circuitpython.org/en/latest/shared-bindings/usb/core/index.html#usb.core.Device.set_configuration) and [usb.core.Device.read()](https://docs.circuitpython.org/en/latest/shared-bindings/usb/core/index.html#usb.core.Device.read) to read from an endpoint on the yubikey? (perhaps endpoint 0x81 if it has that?)

Based on what I've seen from working with usb.host.Device, I would expect there's a good chance that approach might work. But, you would probably need a try/except block around `read()` to deal with USBTimeoutError. You might also need a retry loop around the `find()` call with try/except block for USBError. I've had good luck with that approach for certain USB devices. But, there are also some devices which CircuitPython does like talking to, for reasons I do not understand yet.

Also, have you tried writing a PyUSB driver for your yubikey in desktop Python? That can be a useful way to understand if you are encountering a CircuitPython usb.core implementation problem, or if the problem has more to do with your code not speaking to a USB device in the manner it likes to be spoken to. If you write a PyUSB driver for Python on Linux, it can be pretty helpful to watch what's going on in the USB stack using Wireshark (compare your driver to OS drivers, etc).

p3lim, 2024-10-08:
> Could you give more detail about what you tried, what worked, and what didn't work? Maybe with some sample code?

I was just running the tests found here: https://github.com/adafruit/circuitpython/tree/main/tests/circuitpython-manual/usb

With device_info.py I was able to detect the devices no problem. With basic_keyboard.py I was able to read the OTP event from the yubikey. With basic_mouse.py I could read the vectors from the mouse.

> On the side of CircuitPython acting as USB host, were you trying to use to use [usb.core.find()](https://docs.circuitpython.org/en/latest/shared-bindings/usb/core/index.html#usb.core.find), [...]

Correct.

> Based on what I've seen from working with usb.host.Device, I would expect there's a good chance that approach might work. But, you would probably need a try/except block around read() to deal with [...]

This worked just fine for input HID devices like a keyboard or mouse, but for HID CDC there needs to be support in CircuitPython and/or TinyUSB for passthrough.

> Also, have you tried writing a PyUSB driver for your yubikey in desktop Python? That can be a useful way to understand if you are encountering a CircuitPython usb.core implementation problem, or if [...]

No, I'm not that familiar with the inner workings of CircuitPython, I just use it to write keyboard firmware using the provided APIs.

samblenny, 2024-10-08:
So, is it the case that you just need a way to send usb_hid keystrokes? Can you just buffer what you get from the yubikey, then send it to the host computer using [adafruit_hid.keyboard](https://learn.adafruit.com/circuitpython-essentials/circuitpython-hid-keyboard-and-mouse)?

Are you saying that you've tried doing that and adafruit_hid.keyboard does not work?

Or, if you just want literal pass through with no changes at all, could you just include a hub in your keyboard build (perhaps [Adafruit CH334F Mini 4-Port USB Hub Breakout](https://www.adafruit.com/product/5997))?

p3lim, 2024-10-08:
> So, is it the case that you just need a way to send usb_hid keystrokes? Can you just buffer what you get from the yubikey, then send it to the host computer using [adafruit_hid.keyboard](https://learn.adafruit.com/circuitpython-essentials/circuitpython-hid-keyboard-and-mouse)?

No, I'd need full generic USB passthrough, or at the very least CDC, for the yubikey FIDO2/U2F to work.

> Can you just buffer what you get from the yubikey, then send it to the host computer using [adafruit_hid.keyboard](https://learn.adafruit.com/circuitpython-essentials/circuitpython-hid-keyboard-and-mouse)?
> 
> Are you saying that you've tried doing that and adafruit_hid.keyboard does not work?

Are you saying that you've tried doing that and adafruit_hid.keyboard does not work?

For yubikey OTP or the mouse I could just buffer the input and send it like this and that'd work fine.

> Or, if you just want literal pass through with no changes at all, could you just include a hub in your keyboard build (perhaps [Adafruit CH334F Mini 4-Port USB Hub Breakout](https://www.adafruit.com/product/5997))?

I could do that, but how would that communicate through the RP2040? Or do you suggest that I rather do this?:
```
PC -> HUB -> RP2040
       \---> yubikey
        \--> mouse
```

samblenny, 2024-10-08:
oh... okay, so by passthrough for HID CDC, do you mean you want to bridge a serial port connection between the RP2040's USB Host port and its upstream connection to the computer? In that case, I have no idea if that's possible, or how to go about it. I've never tried to make a USB serial port connection over a CircuitPython USB host port. You could try \#help-with-circuitpython on Adafruit Discord to see if anybody there knows about doing that.

> Or do you suggest that I rather do this?: ...

Yeah, using a physical hub might be a much more straightforward and performant solution if I understand your problem correctly.

p3lim, 2024-10-08:
> oh... okay, so by passthrough for HID CDC, do you mean you want to bridge a serial port connection between the RP2040's USB Host port and its upstream connection to the computer? In that case, I [...]

Not specifically, but it'd be one of the things it'd have to support. I imagine I could use one of the ports for a flash drive in a pinch. This whole thing is about accessibility of USB ports, and portability for devices I usually bring with me to work.

> Yeah, using a physical hub might be a much more straightforward and performant solution if I understand your problem correctly.

I've been trying to do this the other way around (e.g. using the RP2040 as a make-shift hub in addition to its normal duties), but of course it'd be _way_ easier to just add a proper hub before it :facepalm:. Thanks for the suggestion, I'll go that route instead.

closed 2024-10-08 by p3lim