adafruit/circuitpython #9703: RP2350 does not reset i2c peripheral upon reload

bill88t, 2024-10-09:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.0-beta.0-25-g7f14b9c49a on 2024-10-04; Raspberry Pi Pico 2 with rp2350a
```

### Code/REPL

```python
import time, board, busio, adafruit_ssd1306
from digitalio import DigitalInOut
i2c = busio.I2C(board.GP17, board.GP16)
display = adafruit_ssd1306.SSD1306_I2C(128, 64, i2c)
```

### Behavior

SSD1306 display properly attached
First time works, then if we reload and run it again:

ValueError: I2C peripheral in use

tannewt, 2024-10-09:
I don't think this is a bug. We keep I2C around when used by a display so that it can be used internally. Do `displayio.release_displays()` to release it.

bill88t, 2024-10-09:
It's not a displayio display. I just met the bug with it.
It can be repro'ed with just `import board, busio; busio.I2C(board.GP17, board.GP16)`, reloading and running it again.

dhalbert, 2024-10-09:
I tried latest `main` with a SAMD21 board, and I2C cleanup is acting flaky, though I haven't got a reproducer yet. I think this may be due to #9671, which removed `i2c_reset()` on various ports.

PR #9706 (2024-10-10): add __del__() to busio.I2C, which now has a finaliser
- Fixes #9703 

`busio.I2C()` became a class with a finaliser in #9671, but it also needed `__del__()` to  be defined for the finaliser to work properly.

Tested on a Metro RP2040, a PyPortal, and a Pico 2.
closed 2024-10-10 by tannewt