adafruit/circuitpython #9704: Remove `aesio`?

tannewt, 2024-10-09:
Does anyone use `aesio`? It takes about 3k on ESP32-C3. We'll likely add a new crypto API for Matter support. (Currently it uses a small part of `cryptography` and `ecdsa` modules.)

dhalbert, 2024-10-09:
`aesio` was added by #2823. It implies it might have been used for some Covid tracking functionality. @xobs Are you using `aesio` for anything now? You were the PR author.

jepler, 2024-10-09:
ping also to @bobricius  and @isacben who have also previously contributed fixes or improvements to this module

mrdalgaard, 2024-10-10:
@tannewt  I use aesio to secure LoRa packets

isacben, 2024-10-10:
Hi @jepler
Thanks for checking! I am no using it.

dhalbert, 2025-02-01:
I am using it for CircuitMatter on CircuitPython right now.

closed 2025-02-04 by tannewt