adafruit/circuitpython #9710: Raspberry Pi Pico 2 can't enter Safe Mode

Supermath101, 2024-10-11:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.0-beta.0 on 2024-09-17; Raspberry Pi Pico 2 with rp2350a
```

### Code/REPL

```python
# boot.py
import usb_cdc
usb_cdc.disable()
```

### Behavior

After power-cycling the Raspberry Pi Pico 2, as instructed in https://github.com/adafruit/circuitpython/issues/990#issuecomment-444643691, the `boot.py` still runs.

### Description

I was expecting a REPL prompt similar to this, but more applicable to the actual situation:
```
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
The power dipped. Make sure you are providing enough power.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 8.1.0 on 2023-03-30; Adafruit Feather RP2040 with rp2040
```
– https://learn.adafruit.com/circuitpython-safe-mode/overview

However, I was instead met with no REPL, as the `boot.py` disabled the REPL.

dhalbert, 2024-10-11:
I have reproduced this. I'm thinking that the magic word that detects repeated resets is not being saved for some reason (does reset on the RP2350 clear RAM)?

Supermath101, 2024-10-11:
> I have reproduced this. I'm thinking that the magic word that detects repeated resets is not being saved for some reason (does reset on the RP2350 clear RAM)?

If that's actually the situation, maybe safe mode could be entered by pressing BOOTSEL during the  initial 700ms timeframe. At least on the original Pico, the BOOTSEL button could be read, albeit slowly: https://arduino-pico.readthedocs.io/en/stable/bootsel.html

jepler, 2024-10-18:
I investigated this but didn't find a solution

 * I looked for differences between our link files & the official one. There were differences, but trying to harmonize them (where it comes to the uninitialized data section) didn't solve anything
 * Under a picoprobe debugger, if I **use the debugger to reset** (ctrl-c break and then issue the gdb command "run") during the safe mode window, it works
 * In the repl, programmatically resetting to safe mode works, which also goes by the same "saved word" method
 * pressing the feather rp2350 reset button breaks the debugger connection so I can't see what happens in gdb
 * I noticed we have the rp2350 "pico bootsel via double reset" code linking in but eliminating it didn't do anything

I looked for any related issues reported against pico-sdk but came up empty.

eightycc, 2025-04-07:
Using a modified version of the BOOTSEL button sense code works. Pressing BOOTSEL when the LED starts flashing immediately after applying power or pressing RESET (if there is one) successfully enters safe mode.

closed 2025-04-15 by tannewt