adafruit/circuitpython #9726: Pico-W: Setting wifi.radio.dns sends device into safe-mode

bablokb, 2024-10-17:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.0-beta.1 on 2024-10-11; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
#import helpers
#helpers.connect()  connect to AP with wifi.radio.connect
import wifi
wifi.radio.dns
wifi.radio.dns = ('8.8.8.8',)
```

### Behavior

```
>>> import helpers
>>> helpers.connect()
connecting to AP wlan-xxx ...
  connected: False
Unbekannter Fehler 205
  connected: True
>>> import wifi
>>> wifi.radio.dns
('10.1.2.1',)
>>> wifi.radio.dns = ('8.8.8.8',)

[16:30:07.695] Disconnected
[16:30:08.697] Warning: Could not open tty device (No such file or directory)
[16:30:08.697] Waiting for tty device..
[16:30:10.699] Warning: Could not open tty device (Permission denied)
[16:30:10.699] Waiting for tty device..
[16:30:11.703] Connected
Automatisches Neuladen ist deaktiviert.
Sicherheitsmodus aktiv! Gespeicherter Code wird nicht ausgeführt

Sie befinden sich im abgesicherten Modus, weil:
Der CircuitPython-Kerncode ist hart abgestürzt. Hoppla!
Hard fault: memory access or instruction error.
Reiche bitte ein Problem mit deinem Programm bei github.com/adafruit/circuitpython/issues ein.
Drücke Reset, um den Sicherheitsmodus zu beenden.

Drücke eine beliebige Taste um REPL zu betreten. Drücke STRG-D zum neuladen.

Adafruit CircuitPython 9.2.0-beta.1 on 2024-10-11; Raspberry Pi Pico W with rp2040
>>> 
```

### Description

When setting `wifi.radio.dns` the device goes into safe-mode.

BTW: setting with `wifi.radio.dns = ["8.8.8.8"]` is even worse, the system does not even boot and reconnect again.

As a side note: the documentation of `wifi.radio.dns` could be more explicit about the argument: it is currently documented as `Sequence[str]`, but in fact it must be currently a sequence of length zero or one. Also, think about the consistency of the API: all other methods of `wifi.radio` that take an IP as argument expect an instance of `ipaddress.IPv4Address`.

jepler, 2024-10-17:
The reason `wifi.radio.dns` is different is that it can support IPv6 addresses, and is even prepared to support multiple DNS servers when the underlying IP stack supports it as well. That said, it's clearly a bug when setting the property causes a crash.

It would be useful to know if setting the `ipv4_dns` property, which does have to be an IPV4Address, is also crashing; and it might serve as a workaround for you in your code.

jepler, 2024-10-17:
You can also test with an artifact from the actions on #9727 if you have time

bablokb, 2024-10-18:
I will try your patch and report back.

> It would be useful to know if setting the `ipv4_dns` property, which does have to be an IPV4Address, is also crashing; and it might serve as a workaround for you in your code.

Currently I am using `wifi.radio.set_ipv4_address()` which takes IPV4Address arguments. That works fine. I will test the setter too. The main reason I actually tried `wifi.radio.dns` was that I wanted to set more than one DNS, which is not supported as I learned.

bablokb, 2024-10-18:
Setter works fine and #9727 fixes the issue. Thanks!

PR #9727 (2024-10-18): Fix crash on assignment of wifi.radio.dns on raspberrypi boards
This is a speculative fix. I hope that it 

Closes: #9726
closed 2024-10-18 by dhalbert