adafruit/circuitpython #9733: errno.EROFS isn't available but its error string is

ricknun, 2024-10-19:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.4 on 2024-09-17; Adafruit Circuit Playground Bluefruit with nRF52840
```

### Code/REPL

```python
...
import errno
...
try:
    with open("/textfile.txt", "a") as fp:
        ...
except OSError as e:
    print(e)  # for example prints "[Errno 30] Read-only filesystem"
    if e.errno == 30:
    ...
```

### Behavior

The provided code works fine. The "print(e)" results in the string "[Errno 30] Read-only filesystem" and "if e.errno == 30" works as expected. But if I replace "30" with "errno.EROFS" (as described at https://docs.circuitpython.org/en/latest/docs/library/errno.html) I get "During handling of the above exception, another exception occurred" and "AttributeError: 'module' object has no attribute 'EROFS'".

### Description

I would expect to have either both the string and enum (hopefully, since this is such a common error), or neither.

### Additional information

See https://forums.adafruit.com/viewtopic.php?p=1032465 where this issue was discussed with Dan Halbert. Most important, I think the list of strings and list of enums in moderrno.c should agree and both include Read-Only File System. Less important (but it sure would have helped me), the documentation at https://docs.circuitpython.org/en/latest/docs/library/errno.html could be improved from "The codes available may vary per CircuitPython build" (please avoid disclaimers like this) to **where to find the list that applies to my situation**.

ricknun, 2024-10-19:
I just noticed Dan addressed this yesterday. See https://github.com/adafruit/circuitpython/pull/9731.

ricknun, 2024-10-19:
But I don't think Dan's issue https://github.com/adafruit/circuitpython/pull/9731 addresses my request above for improved documentation...

dhalbert, 2024-10-21:
> Less important (but it sure would have helped me), the documentation at https://docs.circuitpython.org/en/latest/docs/library/errno.html could be improved from "The codes available may vary per [...]

I have opened #9737 to cover this.

closed 2024-10-21 by jepler

dhalbert, 2025-03-22:
Further fixed by #10160.

PR #9731 (2024-10-21): Add errno.EROFS and errno.ENOSPC
Motivated by https://forums.adafruit.com/viewtopic.php?t=214213. Fixes #9733

`errno.EROFS` and `errno.NOSPC` didn't exist, but they get nicer error messages already in `mp_common_errno_to_str()`. Add them. A user wanted to check for `errno.EROFS` and was surprised it was not available. `errno.EROFS` is common, given that CIRCUITPY is read-only when connected to USB.

These have not been added to the smallest builds, for space reasons.
PR #10160 (2025-03-21): Fix multiple documentation issues.
- Fixes #10037.
- Fixes #10147.
- Fixes #9690.
- Fixes #9733.
- Fixes #7943.
- Fixes #7743.
- Fixes #4493.

Also fixes a few minor typos I encountered elsewhere.