adafruit/circuitpython #9735: crash when using HTTPServer

dhalbert, 2024-10-20:
@dhalbert Same issue

`Adafruit CircuitPython 9.2.0-beta.1 on 2024-10-11; Maker Go ESP32C3 Supermini with ESP32-C3`

As far as I can tell, `server.poll()` might be the cause.

Currently using the following code.py that still crashes:
```
import wifi
import socketpool
from adafruit_httpserver import Server

pool = socketpool.SocketPool(wifi.radio)
server = Server(pool, debug=False)

while True:
   try:
      server.poll() 

   except Exception as e:
      print(str(e))
```

_Originally posted by @c--- in https://github.com/adafruit/circuitpython/issues/9173#issuecomment-2424948178_

dhalbert, 2024-10-20:
> Is this a duplicate of #9460 ?

It might be but it's a much simpler reproducer, which is great. We should try to fix this and then see ask @mMerlin if #9460 is fixed.

I consider this a showstopper for 9.2.0.

dhalbert, 2024-10-20:
@c--- The test program above generates continuous `ServerStopped` exceptions. This is more easily seen if you do
```py
while True:
   try:
      server.poll() 

   except Exception as e:
      print(type(e))
```
to print the type rather than the string form of the exception. 

You do need a `server.start()`. However, in your [original test program](https://github.com/adafruit/circuitpython/issues/9173#issuecomment-2423923405), you had:
```py
server.start("0.0.0.0", 80)
```
If I add that, I do indeed get a hard crash. But instead of using 0.0.0., use this:
```py
server.start(str(wifi.radio.ipv4_address))
```
In other words, for now, you need to specify the IP address specifically; `"0.0.0.0"` doesn't work. That's a bug we should fix.
You can also use `server.serve_forever(...)`, which does the `.start()` and the `.poll()` loop for you.

mMerlin, 2024-10-21:
What are the implications of this discovery for #9460 which was thought to be the same (or at least related) bug. That does not use the 0.0.0.0 address.

dhalbert, 2024-10-21:
> What are the implications of this discovery for #9460 which was thought to be the same (or at least related) bug. That does not use the 0.0.0.0 address.

I don't think these are the same.

dhalbert, 2024-10-22:
Reproduced, but only on ESP32-C3. Test program:
```py
import wifi
import socketpool
from adafruit_httpserver import Server
import time

time.sleep(5)
print("IP addr", wifi.radio.ipv4_address)

pool = socketpool.SocketPool(wifi.radio)
server = Server(pool, debug=False)

# port 80 will get EADDRINUSE due to web workflow
# Either IP adddress will cause a crash eventually.
server.start("0.0.0.0", 8000)
#server.start(str(wifi.radio.ipv4_address), 8000)

while True:
   try:
      server.poll()
   except Exception as e:
      print(type(e))
```
Either `servert.start()` can cause a crash: the wildcard IP address or the actual IP address don't matter. It takes around 1-5 minutes to restart in safe mode, though I have seen it run longer.

I have reproduced this only on a QT Py ESP32-C3. I cannot reproduce on a QT ESP32-S3, with no PSRAM. I was hoping it might have been a storage leak.

dhalbert, 2024-10-23:
I'm closing this in favor of #9749. There are several red herrings above, which are irrelevant to the actual issue.