adafruit/circuitpython #9736: Documented DVI video modes for RP2350 HSTX Not Implemented in Code

TheKitty, 2024-10-20:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.0-beta.1 on 2024-10-11; Adafruit Feather RP2350 with rp2350a
Board ID:adafruit_feather_rp2350
UID:A5C743E96A479942
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2024 Scott Shawcroft for Adafruit Industries
# Error chercking by Anne Barela
#
# Docs: https://docs.circuitpython.org/en/latest/shared-bindings/picodvi/
#       index.html#picodvi.Framebuffer
#
#       States: color_depth can be 1, 2, 4, 8, 16
#               width can be 320, 400, 640, and 800 pixels
#               height can be 240 ior 480 pixels depending on color_depth
#
# Code at https://github.com/adafruit/circuitpython/blob/main/ports/
#                 raspberrypi/common-hal/picodvi/Framebuffer_RP2350.c
#
# SPDX-License-Identifier: MIT
import sys
import board
import picodvi
import framebufferio
import displayio
import neopixel

displayio.release_displays()

leds = neopixel.NeoPixel(board.NEOPIXEL, 1, brightness=0.4)
leds.fill((0, 255, 0))  # Initial status: green

try:
    fb = picodvi.Framebuffer(400, 240, clk_dp=board.CKP, clk_dn=board.CKN,
                             red_dp=board.D0P, red_dn=board.D0N,
                             green_dp=board.D1P, green_dn=board.D1N,
                             blue_dp=board.D2P, blue_dn=board.D2N,
                             color_depth=8)
except ValueError as e:
    print("Error: ", e)
    # Per picodvi/Framebuffer_RP2350.c only 320x240 at 8 & 16 bits work
    # Show error
    leds.fill((255, 0, 0))  # Error status: red
    sys.exit(e)
    
display = framebufferio.FramebufferDisplay(fb)

# Initialize the display in the display variable
ruler = displayio.OnDiskBitmap("/display-ruler-rgb-720p.bmp")

t = displayio.TileGrid(ruler, pixel_shader=ruler.pixel_shader)

g = displayio.Group()
g.append(t)

display.root_group = g

display.refresh()

while True:
    pass
```

### Behavior

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
]0;🐍code.py | 9.2.0-beta.1\Error:  Invalid width and height
]0;�Done | 9.2.0-beta.1\
Code done running.
NeoPixel red

### Description

When the screen with and height are set to 320 x 200 at 8 or 16 bit color_depth, it works. Any other values give an error. Consistent with https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/common-hal/picodvi/Framebuffer_RP2350.c but inconsistent with what Scott said and https://docs.circuitpython.org/en/latest/shared-bindings/picodvi/index.html#picodvi.Framebuffer

### Additional information

Mismatch between capability in documentation (and as relayed by Scott) and the current code. I'm unsure how easy it would be to add the additional resolutions.

tannewt, 2024-10-21:
Opened #9739 to implement 800x480 output. This will close with doc updates.

TheKitty, 2024-10-21:
@tannewt is 800x480 implemented for RP2350 [per above](https://github.com/adafruit/circuitpython/issues/9736#issuecomment-2427378817_)? I looked at the PR and the code header comments appeared that for RP2350 (vs RP2040) "On RP2350, output resolution is always 640x480"

tannewt, 2024-10-21:
It is not implemented. The issue is to track its implementation.

TheKitty, 2024-10-21:
Cool, thank you so much

PR #9740 (2024-10-21): Document picodvi differences
RP2350 only support 640x480 output currently.

Fixes #9736
closed 2024-10-21 by tannewt