adafruit/circuitpython #9743: ESP32-C3 cannot detect if it is connected to USB.

berkut0, 2024-10-22:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.4
Adafruit CircuitPython 9.2.0-beta.1
```

### Code/REPL

```python
It was written for rp2040-zero and the code works perfectly.
But this code doesn't work for esp32-c3 based boards.

import board, supervisor
import neopixel

pixels = neopixel.NeoPixel(board.GP16, 1, auto_write=True)

while True:
    if supervisor.runtime.usb_connected:
        pixels[0] = (0,255,0)
    else:
        pixels[0] = (255,0,0)
```

### Behavior

The LED should be green when the chip is connected via USB and red when the chip is connected to the power supply. In the case of rp2040-zero everything works perfectly, but esp32-c3 gives a false value all the time.

deshipu, 2024-10-22:
The C3 doesn't have USB support as far as I remember.

berkut0, 2024-10-22:
@deshipu But the chip detects the Serial connection perfectly. So far it looks more like a bug.

deshipu, 2024-10-22:
How does it detect serial connection?

berkut0, 2024-10-22:
@deshipu 

```python
if supervisor.runtime.serial_connected:
    pass
```

deshipu, 2024-10-22:
That seems correct to me? Since the board is using an external USB to serial chip, USB is never connected to the microcontroller itself, so the False response is correct.

deshipu, 2024-10-22:
Quoting from https://circuitpython.org/board/adafruit_qtpy_esp32c3/

> ESP32-C3 is a low-cost microcontroller from Espressif that supports 2.4 GHz Wi-Fi and Bluetooth® Low Energy (Bluetooth LE). It has built-in USB-to-Serial, but not native USB - it cannot act as a keyboard or disk drive. The chip used here has 4MB of Flash memory, 400 KB of SRAM and can easily handle TLS connections.

berkut0, 2024-10-22:
@deshipu Aaah, thank you so much!
That explains everything.

closed 2024-10-22 by dhalbert