adafruit/circuitpython #9746: Update PSRAM cache flush to avoid possible crash under high write workloads

earlephilhower, 2024-10-23:
First of all, thanks for the PSRAM code which we borrowed for [arduino-pico](https://github.com/earlephilhower/arduino-pico)!  It made things much easier to get PSRAM `.heap` and `.data` running for us when the Pico2 came out!

We recently ran into an issue where flash writes caused data loss in PSRAM.  Basically the same two issues you discovered and worked around, we had the pleasure of discovering and working around on our own.  We added the QMI register save/restore and the XIP cache clean process around flash operations and ended up with [essentially your code](https://github.com/adafruit/circuitpython/blob/45e0cf10ccffeb4f5e0bfcb42c8823b648a5f732/ports/raspberrypi/supervisor/internal_flash.c), too.

However, under very dirty cache cleans the processor can repeatably end up crashing during the cleaning process.  The debug is in the RPI Forum thread [here](https://forums.raspberrypi.com/viewtopic.php?t=378249).

Essentially, it seems that the XIP cache really need to be both cleaned and invalidated line-by-line before the flash operations.  This would add another write in your cache clear loop,
https://github.com/adafruit/circuitpython/blob/45e0cf10ccffeb4f5e0bfcb42c8823b648a5f732/ports/raspberrypi/supervisor/internal_flash.c#L53-L55

becomes

````
 for (int i = 1; i < 16 * 1024; i += 8) { 
     maintenance_ptr[i] = 0;  // Clean (write back)
     maintenance_ptr[i - 1] = 0;  // Invalidate
 } 
````

If you do find users with random crashes under PSRAM and flash write workloads, this change might help avoid a lot of head scratching and hair pulling.

Thanks again for the great work!

dhalbert, 2024-10-23:
Thanks @earlephilhower! @tannewt would you like to do a PR?

tannewt, 2024-10-23:
Yes, I'll make a PR. I think I got this code from MicroPython so I'll check there too. :-)

PR #9752 (2024-10-23): Explicitly invalidate cache lines on RP2350
Clean alone can lead to crashes.

Fixes #9746
commit 9bda80b9 (2024-10-23): Explicitly invalidate cache lines on RP2350

Clean alone can lead to crashes.

Fixes #9746
closed 2024-10-23 by dhalbert