adafruit/circuitpython #9762: TTGO T8 V1.7 ESP32 board lacks PSRAM management

robertoseno, 2024-10-27:
Circuitpython releases 9.1.4 and 9.2.0 manage apparently correctly the 4MB Flash but both lack PSRAM management.
You can find in circuitpython-main/ports/espressif/boards/ttgo_t8_v1_7/mpconfigboard.mk that there is no PSRAM declaration:

CIRCUITPY_CREATOR_ID =  0x000C303A
CIRCUITPY_CREATION_ID = 0x00320005

IDF_TARGET = esp32

CIRCUITPY_ESP_FLASH_MODE = qio
CIRCUITPY_ESP_FLASH_FREQ = 80m
CIRCUITPY_ESP_FLASH_SIZE = 4MB

CIRCUITPY_ESPCAMERA = 0

Looking at Adafruit FeatherS2 for example you can see:
USB_VID = 0x239A
USB_PID = 0x80EC
USB_PRODUCT = "Adafruit Feather ESP32S2"
USB_MANUFACTURER = "Adafruit"

IDF_TARGET = esp32s2

CIRCUITPY_ESP_FLASH_MODE = qio
CIRCUITPY_ESP_FLASH_FREQ = 80m
CIRCUITPY_ESP_FLASH_SIZE = 4MB

CIRCUITPY_ESP_PSRAM_SIZE = 2MB
CIRCUITPY_ESP_PSRAM_MODE = qio
CIRCUITPY_ESP_PSRAM_FREQ = 80m

Please can you add PSRAM management for  TTGO T8 V1.7 ESP32 board too?
Thanks

Roberto

dhalbert, 2024-10-27:
Do you feel like you could submit a PR? Most of us don't have this board available to test.

aguaviva, 2024-11-11:
I added support for this board, but I am not sure how to add support for that. Any idea?

aguaviva, 2024-11-23:
I tried adding the following 3 lines:

```
CIRCUITPY_CREATOR_ID =  0x000C303A
CIRCUITPY_CREATION_ID = 0x00320005

IDF_TARGET = esp32

CIRCUITPY_ESP_FLASH_MODE = qio
CIRCUITPY_ESP_FLASH_FREQ = 80m
CIRCUITPY_ESP_FLASH_SIZE = 4MB

CIRCUITPY_ESP_PSRAM_SIZE = 4MB   <--------------------------
CIRCUITPY_ESP_PSRAM_MODE = qio   <--------------------------
CIRCUITPY_ESP_PSRAM_FREQ = 80m   <--------------------------

CIRCUITPY_ESPCAMERA = 0
```

but I get the following warnings:

```
-- Project sdkconfig file /home/raul/repos/circuitpython/circuitpython/ports/espressif/build-ttgo_t8_v1_7/esp-idf/sdkconfig
Loading defaults file /home/raul/repos/circuitpython/circuitpython/ports/espressif/esp-idf-config/sdkconfig-ble.defaults...
Loading defaults file /home/raul/repos/circuitpython/circuitpython/ports/espressif/esp-idf-config/sdkconfig.defaults...
Loading defaults file /home/raul/repos/circuitpython/circuitpython/ports/espressif/esp-idf-config/sdkconfig-opt.defaults...
Loading defaults file /home/raul/repos/circuitpython/circuitpython/ports/espressif/esp-idf-config/sdkconfig-flash-4MB-no-ota-no-uf2.defaults...
Loading defaults file /home/raul/repos/circuitpython/circuitpython/ports/espressif/esp-idf-config/sdkconfig-flash-qio.defaults...
Loading defaults file /home/raul/repos/circuitpython/circuitpython/ports/espressif/esp-idf-config/sdkconfig-flash-80m.defaults...
Loading defaults file /home/raul/repos/circuitpython/circuitpython/ports/espressif/esp-idf-config/sdkconfig-psram.defaults...
Loading defaults file /home/raul/repos/circuitpython/circuitpython/ports/espressif/esp-idf-config/sdkconfig-psram-4MB.defaults...
Loading defaults file /home/raul/repos/circuitpython/circuitpython/ports/espressif/esp-idf-config/sdkconfig-psram-qio.defaults...
warning: unknown kconfig symbol 'SPIRAM_MODE_OCT' assigned to 'n' in /home/raul/repos/circuitpython/circuitpython/ports/espressif/esp-idf-config/sdkconfig-psram-qio.defaults
Loading defaults file /home/raul/repos/circuitpython/circuitpython/ports/espressif/esp-idf-config/sdkconfig-psram-80m.defaults...
warning: unknown kconfig symbol 'SPIRAM_MODE_OCT' assigned to 'n' in /home/raul/repos/circuitpython/circuitpython/ports/espressif/esp-idf-config/sdkconfig-psram-80m.defaults
warning: unknown kconfig symbol 'SPIRAM_SPEED_120M' assigned to 'n' in /home/raul/repos/circuitpython/circuitpython/ports/espressif/esp-idf-config/sdkconfig-psram-80m.defaults
Loading defaults file /home/raul/repos/circuitpython/circuitpython/ports/espressif/esp-idf-config/sdkconfig-esp32.defaults...
warning: unknown kconfig symbol 'SPIRAM_MODE_OCT' assigned to 'n' in /home/raul/repos/circuitpython/circuitpython/ports/espressif/esp-idf-config/sdkconfig-esp32.defaults
warning: unknown kconfig symbol 'SPIRAM_SPEED_120M' assigned to 'n' in /home/raul/repos/circuitpython/circuitpython/ports/espressif/esp-idf-config/sdkconfig-esp32.defaults
Loading defaults file /home/raul/repos/circuitpython/circuitpython/ports/espressif/boards/ttgo_t8_v1_7/sdkconfig...
warning: unknown kconfig symbol 'SPIRAM_MODE_OCT' assigned to 'n' in /home/raul/repos/circuitpython/circuitpython/ports/espressif/boards/ttgo_t8_v1_7/sdkconfig
warning: unknown kconfig symbol 'SPIRAM_SPEED_120M' assigned to 'n' in /home/raul/repos/circuitpython/circuitpython/ports/espressif/boards/ttgo_t8_v1_7/sdkconfig
```

Any idea why?

robertoseno, 2024-11-23:
Hi,I am sorry but I stopped trying for similar reasons.I don’t know internal structure of micropython  tool chain use to build the final file and don’t know where to look at for further critical points.Without adafruit support no hope to solve the problem, at least from my point of view.Best regards and good luck

dhalbert, 2024-11-23:
Did you do `make clean BOARD=ttgo_t8_v1_7` before redoing the build, after adding those lines? It looks like the sdkconfig is inconsistent with other settings.

Or, the esp-idf submodule is out of date. Try updating the usbmodules. I use this alias:
```
alias gitsubupdate='git submodule sync --quiet --recursive && git submodule update --init --filter=blob:none'
```

aguaviva, 2024-11-23:
After a few unsuccessful attempts I deleted everything and started from scratch and same thing. BTW It would help if the sequence of commands to compile the espressif modules where one after the other, otherwise one needs to go fishing around

dhalbert, 2024-11-23:
I just tried this on a clone of `adafruit/circuitpython`, on the tip of `main`. It builds for me both without and without the three `PSRAM` lines in `mpconfigboard.mk`, doing a `make clean` in between.

I'm not sure what to suggest trying next. If you are working on your fork, maybe there is something out of date about it. Try building in a clone of `adafruit/circuitpython` rather than your fork.

I revised the instructions in https://learn.adafruit.com/building-circuitpython/espressif-build slightly yesterday, to update them about python venv's.

robertoseno, 2024-11-23:
Thanks DanI will retry with a fresh clone and following your suggestions

aguaviva, 2024-11-24:
Nice comment about the `venv` I think it is safe to assume people are using venvs these days as it helps keep a clean installation.
Also, I was working on a clone of adafruit/circuitpython, not on main but on the `9.2.1` branch, in Ubuntu 24.04. 

If all compiled fine for you (it worked for me despite of the warnings) please consider merging https://github.com/adafruit/circuitpython/pull/9824

aguaviva, 2024-11-30:
the PR for this has been merged https://github.com/adafruit/circuitpython/pull/9824
please @robertoseno please close this bug

closed 2024-11-30 by jepler