adafruit/circuitpython #9768: Invalid output from json.dump/dumps with IPV4Address

romkey, 2024-10-28:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.0 on 2024-10-28; BLING! with ESP32S3
```

### Code/REPL

```python
import json
import ipaddress

ip = ipaddress.IPv4Address('1.1.1.1')
dict = { 'string': 'this is a string', 'ip': ip }
print(json.dumps(dict))
```

### Behavior

[image]

### Description

calling `json.dump()` or `json.dumps()` with an IPv4Address object produces invalid JSON. The dotted quad should be quoted and isn't.

The problem is that it calls the [print helper for IPv4Address](https://github.com/adafruit/circuitpython/blob/main/shared-bindings/ipaddress/IPv4Address.c#L149), which doesn't check whether the print kind is `PRINT_JSON`.

Not specific to any port.

romkey, 2024-10-28:
[PR 9769](https://github.com/adafruit/circuitpython/pull/9769) should fix this

dhalbert, 2024-11-02:
CPython does something yet different, which we should emulate for compatibility. I would not expect `ipaddress.IPv4Address` to work directly in a dict to convert to json, but `str()` should work.
```
Python 3.12.3 (main, Sep 11 2024, 14:17:37) [GCC 13.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import json
>>> import ipaddress
>>> ip = ipaddress.IPv4Address('1.1.1.1')
>>> ip
IPv4Address('1.1.1.1')           # This is what we should do for v4 and v6 addresses.
>>> dict = { 'string': 'this is a string', 'ip': ip }
>>> print(json.dumps(dict))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.12/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/json/encoder.py", line 200, in encode
    chunks = self.iterencode(o, _one_shot=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/json/encoder.py", line 258, in iterencode
    return _iterencode(o, 0)
           ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/json/encoder.py", line 180, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
TypeError: Object of type IPv4Address is not JSON serializable
>>> dict = { 'string': 'this is a string', 'ip': str(ip) }     # this is ok
>>> print(json.dumps(dict))
{"string": "this is a string", "ip": "1.1.1.1"}
```

jepler, 2024-11-02:
In general, CircuitPython inherits from MicroPython the limitation that use of incorrect types in json.dumps is not diagnosed.

jepler, 2024-11-03:
[transferring some remarks from the linked PR]

This issue has been reported upstream at Micropython with the response that "This was a design decision. It's undefined behaviour if you serialise something that can't be serialised" https://github.com/micropython/micropython/issues/1719 and [documented as well](https://docs.micropython.org/en/latest/genrst/modules.html#json-module-does-not-throw-exception-when-object-is-not-serialisable). we may not be reproducing this portion of the documentation in the circuitpython online docs.

romkey, 2024-11-03:
When I wrote the PR my feeling was that it's good to generate valid JSON from arbitrary objects when possible, but I agree with the argument that you shouldn't serialize it when it can't be properly deserialized.

I really dislike knowingly creating invalid JSON; that just feels like a bad practice.

What do y'all prefer? If you want I'm happy to rewrite the PR to throw the exception that CPython throws and that Dan mentioned, or to do something else.

dhalbert, 2024-11-04:
The problem is that there are many, many print routines for native objects, and ideally they would all raise an exception when asked to print to json. So this is just one among many. Instead of taking up code space, we could consider a more general solution that provides for JSON printing for the few objects that work (ints, floats, bools, strings, dicts, sequences), and automatically forbids other things. But that's a low level fix that probably needs a flag or an optional extra field in the object type info struct.

PR #9789 (2024-11-04): Throw TypeError when json-serializing invalid types
To do this without spending a lot of code, a new type flag is used for this.

This behavior is in line with standard Python.

Closes: #9768
commit 9113e171 (2024-11-04): Throw TypeError when json-serializing invalid types

This behavior is in line with standard Python

Closes: #9768
closed 2024-11-04 by dhalbert