adafruit/circuitpython #9774: Can't perform relative import in the root folder

leogout, 2024-10-30:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.0 on 2024-10-30; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
from .test import test

print('main.py')
```

### Behavior

```
Auto-chargement activé. Copiez ou sauvegardez les fichiers via USB pour les lancer ou démarrez le REPL pour le désactiver.
code.py sortie :
Traceback (appels les plus récents en dernier) :
  Fichier "code.py", ligne 1, dans <module>
ImportError: ne peut importer relativement
Exécution du code terminée.
Appuyez sur n'importe quelle touche pour utiliser le REPL. Utilisez CTRL-D pour relancer.
```

Which translates to : 

```
Auto-reload enabled. Copy or save files via USB to run them or start the REPL to disable it.
code.py output:
Traceback (most recent calls last):
File "code.py", line 1, in <module>
ImportError: cannot import relatively
Code execution finished.

Press any key to use the REPL. Use CTRL-D to reload.
```

### Description

I have the following files structure : 
[image]
In the test folder : 
[image]
And here is the source code for the `main.py` and the `test.py` :
```python
from .test import test

print('main.py')
```

```
print('test.py')
```

### Additional information

The hardware connection is done through USB on Windows 10, the output logs are on COM4, reading them with putty

jepler, 2024-10-30:
The relative import syntax is for modules that are within a package, but the main Python code does not reside "in a package". You get something similar when you try to perform the same action on standard desktop Python (3.11 on my system), though the message is different:
```
$ echo 'from .test import test' > issue9774.py
$ python issue9774.py 
Traceback (most recent call last):
  File "/home/jepler/issue9774.py", line 1, in <module>
    from .test import test
ImportError: attempted relative import with no known parent package
```

jepler, 2024-10-30:
You may mean to write `from test import test` instead

leogout, 2024-10-30:
I should mention that I need relative imports in the root folder for the following reason (hope I'll avoid [the XY problem](https://xyproblem.info/)) :
- I am coding a project on my main computer, targeting a raspberry pi pico
- In this project I want to run unit tests, linting, black, etc. on the source code
- As the main file has to be called `code.py` I can't put it in the root folder of my project or in my PYTHONPATH, because it would shadow the `code` module required by pytest and others
- So I needed to put my sources in a subfolder so that everything works during the devlopment phase
  - Everything is in my src/ directory which isn't in PYTHONPATH, so importing the modules for unit testing works by referring to the files with their full path `src.mymoduleA.mymoduleB`
- I want to be able to copy/paste the files from my project directly to the CIRCUITPY drive without making modifications to the source code, but relative imports won't work

Oh okay thanks @jepler, I did not know that. I'll try to find another solution !

jepler, 2024-10-30:
At least the following 4 names are always supported for the primary code file:
```
            "code.txt", "code.py", "main.py", "main.txt"
```
so maybe one of the .txt files would help with what you're trying to do.

leogout, 2024-10-30:
Well thanks @jepler, you just made my day

closed 2024-10-30 by leogout

mMerlin, 2024-10-30:
> At least the following 4 names are always supported for the primary code file:
> 
> ```
>             "code.txt", "code.py", "main.py", "main.txt"
> ```
> 
> so maybe one of the .txt files would help with what you're trying to do.

I use main.py exactly for that reason. It allows me to run unittests both with CPython on the desktop computer, and in Circuitpython. No complaints about shadowing.