adafruit/circuitpython #9780: Duty cycle modulation for synthio Note object (AKA PWM for commonly used on square wave oscillators)

kevinjwalters, 2024-11-01:
A common modulation techniques for synthesizers is modulation of the duty cycle on one or more oscillators. A simple example of a [triangle wave LFO modulating a square or approximate square wave](https://www.youtube.com/watch?v=XE__0oWOsBM&t=305s). It would be useful to have some efficient way of varying the duty cycle on at least a square wave and having that controlled by the usual myriad of `synthio` options.

### Workarounds
I've been doing this by reading the value from an LFO and manually changing the waveform in python code. It's [not perfect but better than I thought it would be](https://www.youtube.com/watch?v=zAC-lq6M-BA). I raised it as part of [Adafruit Forums:  synthio mysteries - high frequency notes, envelope release, LFO mod of square wave duty cycle (PWM) and lpf making noise](https://forums.adafruit.com/viewtopic.php?p=1034152) and @jepler mentioned another technique using `ulab` to (re)create the arrays using `np.where`.

I did a quick bit of benchmarking and for a 2048 element waveform (fairly large to get a nice sound on the changing duty cycle) it's actually more efficient to change what's needed in python code than make a whole new array in `ulab`.

DETAILS TO FOLLOW.

BTW, I gave up using the value from a `synthio.LFO` object in my code as the update every 256 samples (at 32k = 8ms) was too coarse.

dcooperdalrymple, 2024-11-04:
Hi Kevin, I recommend looking into the `waveform_loop_start` and `waveform_loop_end` properties of a `synthio.Note` object (https://docs.circuitpython.org/en/latest/shared-bindings/synthio/index.html#synthio.Note.waveform_loop_start).

Say you want to have a range of 1% to 99% duty cycle, you could generate a square wave array (bigger will provide better detail as you get closer to 99%) to load into `waveform`. If your array is of 1024 elements, you could get 99% duty cycle with a value of 111 on `waveform_loop_end`  (see `set_duty` below for calculation) and 1% with a value of 1024.

``` python3
import audiobusio
import board
import math
import synthio
import ulab.numpy as np
import time

SIZE = 1024
MIN_DUTY = 1 # percentage
VOLUME = 32000

audio = audiobusio.I2SOut(bit_clock=board.GP0, word_select=board.GP1, data=board.GP2)
synth = synthio.Synthesizer(sample_rate=48000)
audio.play(synth)

waveform = np.concatenate((
    np.full(size_high := math.floor(SIZE * MIN_DUTY / 100), VOLUME, dtype=np.int16),
    np.full(size_low := SIZE - size_high, -VOLUME, dtype=np.int16)
))

note = synthio.Note(
    frequency=220,
    waveform=waveform,
)
synth.press(note)

def set_duty(value :int) -> None: # percentage
    value = min(max(value, MIN_DUTY), 100 - MIN_DUTY)
    note.waveform_loop_end = math.ceil(size_high + size_low * value / (100 - MIN_DUTY * 2))

while True:
    for duty in range(100 - MIN_DUTY, MIN_DUTY, -1):
        set_duty(duty)
        time.sleep(0.01)
    for duty in range(MIN_DUTY, 100 - MIN_DUTY):
        set_duty(duty)
        time.sleep(0.01)
```

Currently, these properties don't support `synthio.BlockInput` to allow them to be controlled by a `synthio.LFO` object, but that's something that could be looked into. However, you mention being able to control them more often than every 256 samples.

If you want more granularity than every 8ms, I recommend looking into a DAC that handle sample rates higher than 32khz. I believe the UDA1334 can do 96kHz which would result in LFO updates every 2.6ms.

jepler, 2024-11-04:
Thanks, I'd forgotten about the waveform start/end offsets. This does seem to be a good way to accomplish modulating the duty cycle and I agree that the idea of having them act as BlockInputs is interesting.

I don't think modifying a property such as `waveform` or `waveform_loop_end` from Python code can give an effective update rate of more than the number of samples generated at one go, because Python code is not being evaluated while samples are being calculated.

kevinjwalters, 2024-11-04:
@dcooperdalrymple I'll give it a go. I had thought about ulab slices with moving around a big array as I think they are implemented with views rather than copies.

For LFOs it would be nice to be able to override the default update rate. I have been running at 64k for both AudioMixer and Synth with PWM audio out so it's not as coarse as the example I cited. I did notice it so perhaps I started with some code at 16k as I was doing more bassy stuff and then noted the issue.

I haven't yet explored I2S but I'll get there eventually. I see there's a [Adafruit I2S Stereo Decoder - UDA1334A Breakout](https://www.adafruit.com/product/3678).

kevinjwalters, 2024-11-04:
Is there a way to modify both `waveform_loop_start` and `waveform_loop_end` atomically? My instinct is to maintain the same sample length if possible.

I just inspected the intial values and end looks wrong?

```
>>> len(waveform) ; (note.waveform_loop_start, note.waveform_loop_end)
1024
(0, 16384)
```

Oh, maybe the `waveform_max_length` is a static value and doesn't represent the actual waveform length?

```
>>> note.waveform_loop_end = 512
>>> (note.waveform_loop_start, note.waveform_loop_end)
(0, 512)
>>> note.waveform_loop_end = 1024
>>> (note.waveform_loop_start, note.waveform_loop_end)
(0, 1024)
>>> note.waveform_loop_end = 1025
>>> (note.waveform_loop_start, note.waveform_loop_end)
(0, 1025)
>>> note.waveform_loop_end = 16384
>>> (note.waveform_loop_start, note.waveform_loop_end)
(0, 16384)
>>> note.waveform_loop_end = 16385
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: waveform_loop_end must be 1-16384
```

@dcooperdalrymple For `set_duty()` which only modifies end should it be more like this?

```
def set_duty2(value):
    value = min(max(value, MIN_DUTY), 100 - MIN_DUTY)
    note.waveform_loop_end = round(100 * size_high / value)
```

dcooperdalrymple, 2024-11-04:
Hi Kevin, I realized this morning that I was thinking about the problem wrong. I've added an updated example within the PR I created last night that demonstrates full PWM control in a much better fashion. https://github.com/adafruit/circuitpython/pull/9788#issuecomment-2454815764

Essentially, you'll just need a standard 50/50 square waveform and create a "window" of half of the waveform data. Then, you can move that "window" around left or right to get different duty cycles. Here's a terrible visualization of that in paint.

[image]

The PR I shared above also includes `synthio.BlockInput` support for these values which is working quite well so far.

Without `synthio.BlockInput` support, it would be more like the following:

``` python3
def set_duty(value: float) -> None:
    value = min(max(value, 0.01), 0.99) # 0.0 and 1.0 will create silence
    index = int((1.0 - value) * SIZE / 2)
    note.waveform_loop_start = index
    note.waveform_loop_end = index + SIZE // 2
```

dcooperdalrymple, 2024-11-04:
Here's the updated example (without `synthio.BlockInput`) using the logic above:

``` python3
import audiobusio
import board
import synthio
import ulab.numpy as np
import time

SIZE = 2048
VOLUME = 32000
SAMPLE_RATE = 48000

audio = audiobusio.I2SOut(bit_clock=board.GP0, word_select=board.GP1, data=board.GP2)
synth = synthio.Synthesizer(sample_rate=SAMPLE_RATE)
audio.play(synth)

waveform = np.concatenate((
    np.full(SIZE // 2, VOLUME, dtype=np.int16),
    np.full(SIZE // 2, -VOLUME, dtype=np.int16)
))

note = synthio.Note(
    frequency=220,
    waveform=waveform,
)
synth.press(note)

def set_duty(value: float) -> None:
    value = min(max(value, 0.001), 0.999) # 0.0 and 1.0 will create silence
    index = int((1.0 - value) * SIZE / 2)
    note.waveform_loop_start = index
    note.waveform_loop_end = index + SIZE // 2

while True:
    for duty in range(1000, 0, -1):
        set_duty(duty / 1000)
        time.sleep(0.01)
    for duty in range(0, 1000):
        set_duty(duty / 1000)
        time.sleep(0.01)
```

dcooperdalrymple, 2024-11-04:
> I just inspected the intial values and end looks wrong?
> 
> ```
> >>> len(waveform) ; (note.waveform_loop_start, note.waveform_loop_end)
> 1024
> (0, 16384)
> ```
> 
> Oh, maybe the `waveform_max_length` is a static value and doesn't represent the actual waveform length?

`waveform_loop_end` defaults to the maximum allowed waveform size within synthio. When synthio iterates over the waveform data, it will loop around at the maximum index of the waveform data before `waveform_loop_end` if it is less than that value. Using the maximum allowed length was a way for us to set a default for the `uint32_t` value that wouldn't affect existing synthio programs which don't utilize this setting.

> >>> note.waveform_loop_end = 16385
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> ValueError: waveform_loop_end must be 1-16384

Waveform data cannot be greater than 16384 `int16_t` elements in length. This `ValueError` is appropriate in this case, but if the aforementioned PR is merged, this error message will no longer occur. In the note output loop, it will still cap `waveform_loop_end` to the maximum index of the waveform data which is limited to 16384.

dcooperdalrymple, 2024-11-04:
The PR (https://github.com/adafruit/circuitpython/pull/9788) has been merged into the main branch. This feature should be available within the "Absolute Newest" downloads within https://circuitpython.org/board/raspberry_pi_pico/ in case you aren't building the firmware yourself.

I've got one last quick update to the example using `synthio.LFO`. To avoid any potential drift between the `waveform_loop_start` and `waveform_loop_end`, it would be better to share a single `synthio.LFO` object between them and use `synthio.Math` to add `SIZE // 2` to `waveform_loop_end`:

``` python3
note.waveform_loop_start = synthio.LFO(
    scale=len(note.waveform) // 4,
    offset=len(note.waveform) // 4,
)
note.waveform_loop_end = synthio.Math(
    synthio.MathOperation.SUM,
    note.waveform_loop_start,
    len(note.waveform) // 2,
    0.0 # 1.0 by default
)
```

Here's a gist of the full example: https://gist.github.com/dcooperdalrymple/90ddfcf5ddf87b6cf370b35b1b213331

Unless there's any else do dive into on this topic, I think we're clear to close this issue, @jepler .

closed 2024-11-08 by jepler

kevinjwalters, 2026-03-25:
@relic-se I'm getting back to some synthio work. Just tested changing the start/end with LFO and works well, thanks.