adafruit/circuitpython #9781: RP2350B: invalid microcontroller.cpu.temperature

bablokb, 2024-11-01:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.0 on 2024-10-28; Pimoroni Pico Plus 2 with rp2350b
```

### Code/REPL

```python
>>> import microcontroller
>>> microcontroller.cpu.temperature
-1160.06
```

### Behavior

see above

### Description

This works fine on a Pico2 with a RP2350A. I don't have any other RP2350B board so I can't verify if it is board specific or due to the B-version of the chip.

jepler, 2024-11-01:
```
 * RP2040, RP2350 QFN-60: User ADC inputs are on 0-3 (GPIO 26-29), the temperature sensor is on input 4.
 * RP2350 QFN-80 : User ADC inputs are on 0-7 (GPIO 40-47), the temperature sensor is on input 8.
```
```c
/**
 * The ADC channel number of the on-board temperature sensor
 */
#ifndef ADC_TEMPERATURE_CHANNEL_NUM
#define ADC_TEMPERATURE_CHANNEL_NUM (NUM_ADC_CHANNELS - 1)
#endif
```
Looks like the hard-coded channel value in common_hal_mcu_processor_get_temperature should be changed to use this define instead.

Feel like you could pick this up and PR it? I don't have a 2350B handy to check if it works.

bablokb, 2024-11-02:
Yes, I will do that. Won't be before Tuesday though.

PR #9784 (2024-11-02): use ADC_TEMPERATURE_CHANNEL_NUM instead of hard-coded channel '4'
Fixes #9781 

Tested with
  - Pico (rp2040)
  - Pico2 (rp2350a)
  - Pimoroni Pico Plus2 (rp2350b)
closed 2024-11-02 by dhalbert