adafruit/circuitpython #9790: ESPnow long range mode selection fails

stanelie, 2024-11-04:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.4 on 2024-09-17; Waveshare ESP32-S3-Zero with ESP32S3
```

### Code/REPL

```python
e = espnow.ESPNow(250,31)
```

### Behavior

When selecting integer 31 upon creation of the ESPNow object (which should be selecting line 626 here https://github.com/espressif/esp-idf/blob/a9d0f22193acdf47a5a4db36832ae7068818962b/components/esp_wifi/include/esp_wifi_types.h#L626), communication fails between 2 Waveshare ESP32-S3-zero devices. In fact, electing any mode between 16 and 32 fails.

stanelie, 2024-11-04:
In arduino, I think the correct method to set long range mode is this : 
`esp_wifi_set_protocol(
          ESP_IF_WIFI_STA,
          WIFI_PROTOCOL_LR));`

Is this exposed method in Circuitpython?

anecdata, 2024-11-04:
I don't find `esp_wifi_set_protocol` in the current core implementation, and it's not exposed to Python code as a CircuitPython API, so default of b/g/n is used. For now, I would think 1Mbps would be best for range.

I think the rates 0x10-0x1F require 802.11ax, which is only supported so far in the WiFi-6 ESP32 variants: C5, C6.

stanelie, 2024-11-04:
I would not know where to start to build it myself.
For now, it looks like this has not been implemented in CircuitPython. I've switched to Micropython and was able to make it work there.

Micropython code for reference, with initialization sequence as per the Micropython documentation : 
```
import network
import espnow

sta = network.WLAN(network.STA_IF)
network.phy_mode(network.MODE_LR)
sta.active(True)
sta.disconnect()

e = espnow.ESPNow()
e.active(True)
```

Not sure if this actually works, my range test is inconclusive, but at least, I don't get any error messages from the IDE.
Maybe this can be useful for the CircuitPython people in implementing this feature?

stanelie, 2024-11-05:
Here is un updated version that looks more correct (in how it behaves on my hardware at least) : 
```
import network
import espnow

sta = network.WLAN(network.STA_IF)
sta.active(True)
sta.config(protocol=network.MODE_LR)
sta.config(pm=sta.PM_NONE)
sta.disconnect()

e = espnow.ESPNow()
e.active(True)
```

stanelie, 2024-11-07:
Well,
After further testing, the micropython implementation is also broken, it doesn't work. I've tried all manner of permutation, setting `sta.config(protocol=network.MODE_LR)` before or after `sta.active(True)`, using `network.phy_mode(network.MODE_LR)` instead... 

In all my messing, I was able to set the LR mode once, and I can't figure out how. I was not able to get out of it without re-flashing the ESP32. It looks like this setting "sticks" in the firmware.

I leave this here for reference, in case someone else is fighting with this issue.

anecdata, 2024-11-07:
I'd suggest asking on the MicroPython Discord server, and if that doesn't pan out, file an issue on the MicroPython Gituhb repo.

stanelie, 2024-11-07:
Done : https://github.com/micropython/micropython/issues/16179

stanelie, 2024-11-07:
Well,
After further testing, it looks like even if the LR mode cannot be selected, the default mode used with Circuitpython is still better then what I get with Micropython. I get about twice the range. So, I'll stick with Circuitpython for now, and hope the long range mode can be enabled soon.

anecdata, 2024-11-07:
The ESP-NOW feature was added by a community member, so there's no guarantee it will get an enhancement any time soon. If you are inclined to develop the enhancement yourself, you can get help on Adafruit's Discord server. You could even prove it out first using Espressif's esp-idf environment, with their example.

stanelie, 2024-11-07:
Sorry, I won't be able to do it, I'm just a simple user trying to make a device do what I want it to do, I'm not a developer. I was trying to avoid using Arduino and having to wait for a build everytime I change a coma, because I don't really know how to write code. Circuitpython or micropython looked attractive to me because of this.