adafruit/circuitpython #9808: Add board port for Pimoroni Plus Pico 2 W

eightycc, 2024-11-13:
CP supports the Pimoroni Pico Plus 2 board but lacks support for the new W variant. The Pico Plus 2 W includes a Raspberry Pi RM2 adding WiFi and Bluetooth capabilities that should be compatible with Infineon CYW43439 on the Raspberry Pi Pico W boards.

I'll claim this issue and will submit a pull request as soon as my test board arrives.

anecdata, 2024-11-13:
Related: https://github.com/adafruit/circuitpython/issues/9777

bablokb, 2024-11-14:
I already have (two) of these boards. If I can help in any way, just let me know.

eightycc, 2024-11-14:
Thanks to @anecdata for pointing to https://github.com/micropython/micropython/pull/16057 where a similar Micropython port is underway. Making a first pass through the PR and the changes it makes I can see that RP2350/RM2 changes depend on SDK changes to its `develop` branch. Likewise, cyw43-driver module will also need an update.

It's probably wise to wait for this PR to shake out.

ZodiusInfuser, 2024-11-22:
@bablokb I do not, but I have just put a request in for it, and our other recent boards too. `Pimoroni Pico Plus 2 W` is correct

ZodiusInfuser, 2024-11-22:
Okay. VID/PID available at: https://github.com/raspberrypi/usb-pid

bablokb, 2024-11-22:
@ZodiusInfuser: thanks, that was fast!

I updated my tree.

@eightycc : you claimed this issue, will you create the PR or should I just go ahead?

eightycc, 2024-11-24:
@bablokb Sorry about that. My boards arrived today so I tested and made a PR without first reading your updates.

I explicitly set the PIO clock divider to 2 (also the SDK default) with a note to increase it if the gSPI connection to the RM2 becomes unstable. I don't think it will.

PR #9826 (2024-11-26): Support Pimoroni Pico Plus 2W
Implement support for Pimoroni Pico Plus 2W board vid=0x2E8A pid=0x10bd. Resolves #9808.

Tested with code from "Quick-Start the Pico W WiFi with CircuitPython".
closed 2024-11-26 by tannewt