adafruit/circuitpython #9811: wifi station fails if no access point present, and does not retry

stanelie, 2024-11-15:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.0 on 2024-10-28; Waveshare ESP32-S3-Zero with ESP32S3
```

### Code/REPL

```python
import wifi
import time

SSID = "potato"
PASSWORD = "123123123"

print("connecting wifi...")
wifi.radio.connect(SSID,PASSWORD)
```

### Behavior

If I run this sample with the access point turned off, it errors out with this message and quits to the console, without retrying : 
ConnectionError: No network with that ssid

### Description

If I try wo catch the error and retry, it also fails, before the error catching mechanism kicks in, with the same error message.
```
import time
import wifi

# Replace with your SSID and password
SSID = "potato"
PASSWORD = "123123123"

def connect_to_wifi():
    print("Connecting to AP...")
    wifi.radio.connect(SSID, PASSWORD)

while not wifi.radio.connected:
    try:
        connect_to_wifi()
    except RuntimeError as e:
        print(f"Connection failed: {e}")
        print("Retrying in 5 seconds...")
        time.sleep(5)

if wifi.radio.connected:
    print("Connected to WiFi!")
else:
    print("Failed to connect to WiFi")
```

stanelie, 2024-11-15:
I guess I fixed it with some code that looks over-complicated? Is this the correct way? There is no built-in way?
```
import wifi
import time

SSID = "potato"
PASSWORD = "123123123"

def scan_for_wifi():
    networks = []
    for network in wifi.radio.start_scanning_networks():
        networks.append(network)
    wifi.radio.stop_scanning_networks()
    networks = sorted(networks, key=lambda net: net.rssi, reverse=False)
    for network in networks:
        print("ssid:",network.ssid, "rssi:",network.rssi)
        if network.ssid == SSID:
            print(f"network '{SSID}' is present!")
            return True
        
def connect_to_wifi():
    print("connecting wifi...")
    wifi.radio.connect(SSID,PASSWORD)

    
while True:
    if wifi.radio.connected == True:
        gateway_address = wifi.radio.ipv4_gateway
        ping = wifi.radio.ping(ip=gateway_address)
        if ping is None:
            print(f"Couldn't ping gateway '{gateway_address}' successfully")
        else:
            print(f"Pinging '{gateway_address}' took: {ping * 1000} ms")
        time.sleep(1)
    
    else:
        print(f"scanning for wifi '{SSID}'...")
        scan_for_wifi()
        if scan_for_wifi():
            connect_to_wifi()
        time.sleep(1)
```

dhalbert, 2024-11-15:
`ConnectionError` is not a subclass of `RuntimeError`. So your original code in the first post, will not catch the `ConnectionError` raised inside `connect_to_wifi()`.
```py
    try:
        connect_to_wifi()
    except RuntimeError as e:
```
Try:
```py
    try:
        connect_to_wifi()
    except ConnectionError as e:
```

stanelie, 2024-11-15:
Ah! Sorry, I'm a newbie. Thanks!!

anecdata, 2024-11-15:
@stanelie just fyi... `connect()` does a scan under the hood when it's trying to find the target SSID, and for many connection error types it will automatically retry several times before raising an exception to the user

stanelie, 2024-11-15:
Thanks for the info!
Is this documented somewhere? I'm not sure how to search for this type of info.

Thanks again!

anecdata, 2024-11-15:
It's kind of implementation minutae, may be different in the raspberry pi port, and I suppose low-level behavior could even change in a future version (the core wifi code relies on Espressif APIs). I typically check the code for details like that.