adafruit/circuitpython #9816: ESPNow recovery from "invalid buffer"

stanelie, 2024-11-18:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.0 on 2024-10-28; Waveshare ESP32-S3-Zero with ESP32S3
```

### Code/REPL

```python
def read_packet():
    global packet
    try:
        packet = e.read()
    except ValueError as error:
        print(f"{error}")
```

### Behavior

Hello.
This is more a support question then a bug report. I cannot find anywhere in the documentation how to purge the ESPNow buffer, or how to destroy the ESPNow object so that I can recreate it (to recover from errors).

I run this code where a ESP32 is receiving packets from 2 different senders, and I guess the packets interfere when they arrive at the same time? When that happens, I receive an Invalid buffer error and I cannot recover from this. I can catch the error so the code keeps going, but the buffer keeps saying it is invalid.

anecdata, 2024-11-18:
@stanelie you can try deinit (or as a workaround reload or reset if that doesn't work):
https://docs.circuitpython.org/en/latest/shared-bindings/espnow/index.html#espnow.ESPNow.deinit

stanelie, 2024-11-19:
deinit works perfectly for my purpose. Thanks @anecdata !

jepler, 2024-11-19:
However, I'm glad you were able to work around the problem with the use of `deinit`.

dhalbert, 2024-11-19:
> Actually, now that I think about it `ringbuf` is a CircuitPython API and at is not multithread / multicore safe, so it may not be appropriate to use here at all. For instance, if get and put are happening from multiple cores at the same time, the `used` field (which is not volatile and is simply incremented and decremented with ++ and --) could get garbled.

I think the original, original ringbuf may have been atomically safe, but once we went to two-byte entries, I think that went out the window. That's why I made the implementation more straightforward; there were many latent bugs having to do with the requested size vs needed size of the buffer (one entry more). My current expectation is that ringbuf operations will be done in a critical section.

stanelie, 2024-11-19:
This high level discussion is way above my level, but if you do implement a fix, I will be ready to check if it works.
:)