adafruit/circuitpython #9820: Add STM32U083 support.

Vincent1-python, 2024-11-20:
Add STM32U083 support.

dhalbert, 2024-11-20:
The STM32U083 has 40kB RAM and 256kB flash. The SAMD21, which was our first ports, is 32kB/256kB. The SAMD21 is quite limited in what it can provide as a CircuitPython chip, and so the STM32U083 would be only slightly better. I think it's not so likely we would do this port, given that there are more capable chips now at low cost. But we'd be happy to take a community contribution. Is there some feature of this chip that's particularly interesting?

Vincent1-python, 2024-11-21:
STM32U083
STM32U083x series provides Flash memory up to 256 KB and SRAM up to 40 KB, and it has eight packaging methods, including 32-81-pin UFQFPN, WLCSP, LQFP and UFBGA packages.  
STM32U083x devices provide:
A 12-bit ADC, two built-in rail-to-rail analog comparators and an operational amplifier.
A low-power RTC, a universal 32-bit timer, a 16-bit PWM timer specially used for motor control, three universal 16-bit timers and three 16-bit low-power timers.
Up to 21 capacitive sensing channels
Three I2C's, two SPI's, four USART's and two low-power UART's, and a crystal-free oscillator USB2.0 supporting device mode.
An LCD controller with internal boost converter can drive 8x48 or 4x52 segment codes.
In addition, STM32U083xC device has built-in AES hardware accelerator.

Vincent1-python, 2024-11-21:
Can you add STM32H7R7/S7 support,I think so.

dhalbert, 2024-11-21:
There are already two boards with STM32H7 chips that we support: `nucleo_h743zi_2` and `openmv_h7`.

Vincent1-python, 2024-11-22:
The performance of STM32H7R7/S7 is interesting.

dhalbert, 2024-11-22:
I'm going to close this now, since it seems like you're flexible about which chip to choose. There are many other chips we support as well, with varying levels of performance, besides STM32, if those are of interest.