adafruit/circuitpython #9821: pull-up, pull-down and drive mode for pwmio?

stanelie, 2024-11-20:
### CircuitPython version

```python
Maybe this is simply a documentation problem(or I just cannot read), but how can I set the output mode of a pin I set to use as a PWM output with pwmio? How do I set a pull-up on it, or a pullpdown? How about setting the mode to push-pull or open drain?

Thanks!
```

### Code/REPL

```python
N/A
```

### Behavior

N/A

jepler, 2024-11-20:
I think you're right that this isn't possible at present, so I've marked this issue as an Enhancement.

dhalbert, 2024-11-25:
This might be done by a revamp of `Pin` to include persistent settings that could be changed before a pin is claimed for a `*io` use.