adafruit/circuitpython #9822: types-circuitpython still on version 8.1.0 in PyPI

tekktrik, 2024-11-20:
### CircuitPython version

```python
N/A
```

### Code/REPL

```python
N/A
```

### Behavior

N/A

### Description

`types-circuitpython`, created from type annotations to the code base, hasn't been updated on PyPI since May 2023 (8.1.0 release, 8.2.0b0 pre-release).  It's likely a bug in the CI unless this package is no longer supported.  Happy to dig into it, but wanted to submit a report so I didn't forget!

dhalbert, 2024-11-21:
`types-circuitpython` is not our pypi package. It's somebody else's: https://github.com/hardfury-labs/types-circuitpython.

Are we talking about https://pypi.org/project/circuitpython-stubs/ ? That _is_ up to date.

tekktrik, 2024-11-25:
You are correct, I was referring to the incorrect package.  In that case there isn't any issue, sorry!

closed 2024-11-25 by tekktrik