adafruit/circuitpython #9823: Add explicit pull choice for PinAlarm

dhalbert, 2024-11-22:
See https://forums.adafruit.com/viewtopic.php?t=215000 for motivation.

Currently, if `pull=True` when creating a `PinAlarm`, the pull chosen is the opposite from the `value` argument. From the [doc](https://docs.circuitpython.org/en/latest/shared-bindings/alarm/pin/index.html#alarm.pin.PinAlarm):
> pull ([bool](https://docs.python.org/3/library/functions.html#bool)) – Enable a pull-up or pull-down which pulls the pin to the level opposite that of `value`. For instance, if `value` is set to `True`, setting `pull` to `True` will enable a pull-down, to hold the pin low normally until an outside signal pulls it high.

A user is using a normally-closed switch, connected to a `PinAlarm` pin. (In this case, the switch is closed when a box lid is closed.) The switch is attached to ground at the far end. So, in the sleeping case (lid closed, switch closed) the alarm pin would be low. When the switch opens, the user wants a wake-up.

I didn't consider normally-closed switches when designing the API. If I had, I might have made `pull` take either `None` or `digitalio.Pull.UP` or `digitalio.Pull.DOWN`. However, an upward-compatible addition to the API would be:

- If `pull` is `False` or `None` (`None` would be a new allowed arg), do not enable a pull.
- if `pull` is `True`, enable a `pull` opposite to `value`, as is currently done.
- If `pull` is a `digitalio.Pull`, enable that specific pull.

(The same thing might be done with `keypad`, but we haven't had a call for that yet, and that complicates thinking about "pressed" and "released". So I'm not going there for `keypad`.)

mcmanigle, 2024-12-05:
Commenting to follow. (I was the guy in the original forum post.)

dhalbert, 2025-02-01:
This could be subsumed by #9845.