adafruit/circuitpython #9837: Pico-W: bad network throughput running as a webserver

bablokb, 2024-11-27:
### CircuitPython version

```python
at least since 8.0.5
```

### Code/REPL

```python
see https://github.com/bablokb/circuitpython-examples/tree/master/webserver
```

### Behavior

see below

### Description

The example scenario runs a webserver. The main page sources a number of js and css-files. This triggers multiple parallel requests from the browser to the server. Total download size is about 140KB, throughputs (measured as total download time from within the browser development tools) are as follows:

  - Pico-W (8.0.5 and 9.2.1): about 11s
  - Pico Plus2 W: about 11s
  - Adafruit Feather ESP32-S3 (4MB/2MB): about 1s

### Additional information

This issue is opened on request of @eightycc, see https://github.com/adafruit/circuitpython/pull/9826#issuecomment-2500759235

Searching the web for throughput figures for the Pico-W does not turn up many results. Some tests use iperf, but this is an artificial tool and does not test the given scenario. But maybe iperf could help to track down the cause. It seems there is a Python/MicroPython version available, so it might be worth a look.

bablokb, 2024-11-27:
See this post: https://forums.raspberrypi.com/viewtopic.php?t=339512

They claim using iperf they see

> near 10 Mbits/sec at data size greater than 1 MByte

But this is with the Pico-W running as a client and it is a huge data size for an MCU, so this is not realistic in any way.

eightycc, 2024-11-28:
@bablokb Thank you for opening the issue and providing the diagnostic. I was able to easily reproduce the performance glitch.

Capturing a trace with WireShark, I can see that TCP  PSH/ACK packets from the server are sometimes delayed by up to 500 ms. Interestingly, the delayed packets always arrive on exact 500 ms offsets from the start of the trace, i.e., +4.620 ms, +5.120 ms, and so on. This tells me that ACKs from the client aren't always causing an IRQ or the IRQ is getting ignored. The reason we're beating at 500 ms is that a periodic timer in CircuitPython (or less likely the CYW43) is causing a poll of pending interrupts to occur.

bablokb, 2024-11-30:
I ported iperf to CircuitPython: https://github.com/bablokb/circuitpython-iperf. Might be useful for anybody tweaking network code.

Still needs some polishing and it is currently only tested with a Pico-W. The results reproduce what we already know:

```
> iperf3 -l 4K -c 192.xxx.x.xxx
Connecting to host 192.xxx.x.xxx, port 5201
[  5] local 192.xxx.x.xxx port 35514 connected to 192.xxx.x.xxx port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   792 KBytes  6.49 Mbits/sec    0   27.1 KBytes       
[  5]   1.00-2.00   sec   600 KBytes  4.91 Mbits/sec    0   27.1 KBytes       
[  5]   2.00-3.00   sec   640 KBytes  5.24 Mbits/sec    0   27.1 KBytes       
[  5]   3.00-4.00   sec   600 KBytes  4.92 Mbits/sec    0   27.1 KBytes       
[  5]   4.00-5.00   sec   720 KBytes  5.90 Mbits/sec    0   27.1 KBytes       
[  5]   5.00-6.00   sec   680 KBytes  5.57 Mbits/sec    0   27.1 KBytes       
[  5]   6.00-7.00   sec   640 KBytes  5.24 Mbits/sec    0   27.1 KBytes       
[  5]   7.00-8.00   sec   680 KBytes  5.57 Mbits/sec    0   27.1 KBytes       
[  5]   8.00-9.00   sec   600 KBytes  4.92 Mbits/sec    0   27.1 KBytes       
[  5]   9.00-10.00  sec   680 KBytes  5.57 Mbits/sec    0   27.1 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  6.48 MBytes  5.43 Mbits/sec    0   sender
[  5]   0.00-10.41  sec  6.38 MBytes  5.14 Mbits/sec        receiver

iperf Done.

> iperf3 -l 4K -R -c 192.xxx.x.xxx
Connecting to host 192.xxx.x.xxx, port 5201
Reverse mode, remote host 192.xxx.x.xxx is sending
[  5] local 192.xxx.x.xxx port 57898 connected to 192.xxx.x.xxx port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.02   sec  60.0 KBytes   483 Kbits/sec                  
[  5]   1.02-2.01   sec  52.0 KBytes   432 Kbits/sec                  
[  5]   2.01-3.04   sec  52.0 KBytes   411 Kbits/sec                  
[  5]   3.04-4.03   sec  52.0 KBytes   433 Kbits/sec                  
[  5]   4.03-5.01   sec  52.0 KBytes   433 Kbits/sec                  
[  5]   5.01-6.01   sec  52.0 KBytes   426 Kbits/sec                  
[  5]   6.01-7.09   sec  56.0 KBytes   425 Kbits/sec                  
[  5]   7.09-8.01   sec  48.0 KBytes   426 Kbits/sec                  
[  5]   8.01-9.12   sec  36.0 KBytes   267 Kbits/sec                  
[  5]   9.12-10.19  sec  32.0 KBytes   244 Kbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.47  sec   498 KBytes   389 Kbits/sec    0   sender
[  5]   0.00-10.19  sec   492 KBytes   396 Kbits/sec        receiver

iperf Done.
```

The first run is my laptop sending data, the second run (with the `-R` i.e. reverse mode option) the Pico-W is sending and the laptop is the receiver.

eightycc, 2025-02-01:
The version of LWIP used by RP2 builds hangs up internally when it thinks its statically allocated pcb pool or pbuf pool is exhausted. The hang-up manifests itself as a memory error returned by `tcp_write`. Socket code in CP waits 50 ms and then retries the write. This retry/fail condition persists until LWIP's `tcp_slowtmr` fires and LWIP clears whatever internal condition made it think a pool was exhausted and continues operating normally for a few frames. `tcp_slowtmr` is a 500 ms periodic timer, so that's why the Wireshark trace captures show that heartbeat.

Increasing the size of the pcb and pbuf pools cuts the frequency of the hang-up but does not eliminate it. With both pool sizes increased the time required to serve the webpage in the original report drops from 11 seconds to 2.5 seconds.

Two things need to be done: (1) Tune static memory allocations for LWIP and (2) hunt down the cause of spurious pool full conditions in LWIP and the subsequent hang-ups. The situation is complicated by CP using a locally-modified version of LWIP that is 2 years and >150 pulls behind the main line. 

As a next step, I'm going to rebase the local changes and then hunt down any remaining problems in LWIP related to this issue.

eightycc, 2025-02-01:
#7656 may be related.

eightycc, 2025-02-02:
Re-based local modifications to the top of LWIP's main line. Performance problems still persist. Preliminary list of bugs in LWIP affecting CP:

- When a connection is first established on a TCP socket, scrambled code in `tcp_input` erroneously detects an application rejection of the first frame. This is due to bad arguments sent via the macro `TCP_EVENT_RECV`. The result is a delayed ACK, sometimes delayed by several hundred milliseconds. Eventually LWIP will deliver the frame and send the ACK, but by then the client may have re-sent the un-ACKed frame. Eventually things smooth out, but at the cost of hundreds of milliseconds. Rinse and repeat for every new connection.
- LWIP is configured to use a pool of statically allocated pbufs. However, tracing shows that it is allocating pbufs out of its own tiny heap. As network activity occurs, LWIP's heap becomes fragmented leading to "memory errors". Handling of these errors leads to delays and network performance degrades as an app runs.

eightycc, 2025-02-02:
When a transmission is sent to a TCP socket that is smaller than the TX buffer fill level required to trigger transmission, it will linger in the buffer until a timeout occurs. This imposes a performance penalty on small transmissions and possibly also on the final part of a larger transmission. I'm thinking that this is necessary to avoid sending small fragments to LWIP, but perhaps it can be better tuned by looking at time since last fill in addition to TX fill level.

eightycc, 2025-02-02:
Here is a trace showing the first LWIP bug described above.

**Client (192.168.1.217:42582) sends a SYN packet to begin 3-way TCP handshake:**
<details>
<code>
37.179.473(00.047.302): cyw43_ll_process_packets
37.179.504(00.000.031): cyw43_read_reg_u16 BUS_FUNCTION 0x4=0xbe2020
37.179.565(00.000.061): cyw43_write_reg_u16 BUS_FUNCTION 0x4=0x2020
37.179.595(00.000.030): cyw43_read_reg_u32 BUS_FUNCTION 0x8=0x20928
37.179.626(00.000.031): bus_gspi_status 0x20928 0x104
37.179.656(00.000.030): cyw43_read_bytes 2 0x0 260
37.179.718(00.000.062): [   74359] ETHRX itf=w0 len=74 src=00:1b:21:f0:77:ee dst=28:cd:c1:ff:bf:1d subtype=0800
37.179.870(00.000.152):     IPv4 tlen=60 id=7479 flags=2 frag_offset=0 protocol=6 src=192.168.001.217 dst=192.168.001.230
37.179.901(00.000.031):     TCP src=42582 dst=80 seq=4dacf889 ack=00000000 flags=a002
37.179.931(00.000.030):     0x0000: 28cdc1ff bf1d001b 21f077ee 08004500 : 003c1d37 40004006 9875c0a8 01d9c0a8 : (.......!.w...E. .<.7@.@..u......
37.180.023(00.000.092):     0x0020: 01e6a656 00504dac f8890000 0000a002 : faf0555d 00000204 05b40402 080ae1fc : ...V.PM......... ..U]............
37.180.145(00.000.122):     0x0040: a3c80000 00000103 0307                                                    : ..........
37.180.206(00.000.061): pbuf_alloc(length=74)
37.180.236(00.000.030): pbuf_alloc(length=74) == 0x20025eec
37.180.267(00.000.031): ethernet_input: dest:28:cd:c1:ff:bf:1d, src:00:1b:21:f0:77:ee, type:800
37.180.297(00.000.030): pbuf_remove_header: old 0x20025efc new 0x20025f0a (14)
37.180.328(00.000.031): ip_input: iphdr->dest 0xe601a8c0 netif->ip_addr 0xe601a8c0 (0x1a8c0, 0x1a8c0, 0xe6000000)
37.180.358(00.000.030): ip4_input: packet accepted on interface w0
37.180.389(00.000.031): ip4_input: 
37.180.389(00.000.000): IP header:
37.180.389(00.000.000): +-------------------------------+
37.180.419(00.000.030): | 4 | 5 |  0x00 |        60     | (v, hl, tos, len)
37.180.450(00.000.031): +-------------------------------+
37.180.450(00.000.000): |     7479      |010|       0   | (id, flags, offset)
37.180.480(00.000.030): +-------------------------------+
37.180.480(00.000.000): |   64  |    6  |    0x9875     | (ttl, proto, chksum)
37.180.511(00.000.031): +-------------------------------+
37.180.511(00.000.000): |  192  |  168  |    1  |  217  | (src)
37.180.541(00.000.030): +-------------------------------+
37.180.541(00.000.000): |  192  |  168  |    1  |  230  | (dest)
37.180.572(00.000.031): +-------------------------------+
37.180.572(00.000.000): ip4_input: p->len 60 p->tot_len 60
37.180.603(00.000.031): pbuf_remove_header: old 0x20025f0a new 0x20025f1e (20)
37.180.633(00.000.030): TCP header:
37.180.633(00.000.000): +-------------------------------+
37.180.664(00.000.031): |    42582      |       80      | (src port, dest port)
37.180.664(00.000.000): +-------------------------------+
37.180.694(00.000.030): |           1303181449          | (seq no)
37.180.694(00.000.000): +-------------------------------+
37.180.694(00.000.000): |           0000000000          | (ack no)
37.180.725(00.000.031): +-------------------------------+
37.180.755(00.000.030): | 10 |   |000010|     64240     | (hdrlen, flags (SYN 
37.180.786(00.000.031): ), win)
37.180.786(00.000.000): +-------------------------------+
37.180.786(00.000.000): |    0x555d     |         0     | (chksum, urgp)
37.180.816(00.000.030): +-------------------------------+
37.180.847(00.000.031): pbuf_remove_header: old 0x20025f1e new 0x20025f46 (40)
37.180.877(00.000.030): tcp_input: packed for LISTENing connection.
37.180.908(00.000.031): TCP connection request 42582 -> 80.
37.180.938(00.000.030): tcp_parseopt: MSS
37.180.969(00.000.031): tcp_parseopt: other
37.180.969(00.000.000): tcp_parseopt: other
37.180.969(00.000.000): tcp_parseopt: NOP
37.180.999(00.000.030): tcp_parseopt: other
37.180.999(00.000.000): tcp_enqueue_flags: queuelen: 0
37.181.030(00.000.031): pbuf_alloc(length=4)
37.181.060(00.000.030): pbuf_alloc(length=4) == 0x20028fd8
37.181.091(00.000.031): pbuf_add_header: old 0x20029020 new 0x2002900c (20)
37.181.121(00.000.030): tcp_enqueue_flags: queueing 6510:6511 (0x12)
37.181.152(00.000.031): tcp_enqueue_flags: 1 (after enqueued)
37.181.182(00.000.030): tcp_output: snd_wnd 64240, cwnd 1, wnd 1, effwnd 0, seq 6510, ack 6510
37.181.213(00.000.031): tcp_output: snd_wnd 64240, cwnd 1, wnd 1, effwnd 0, seq 6510, ack 6510, i 0
37.181.243(00.000.030): tcp_output_segment: rtseq 6510
37.181.274(00.000.031): tcp_output_segment: 6510:6510
37.181.304(00.000.030): pbuf_add_header: old 0x2002900c new 0x20028ff8 (20)
[... 461 lines cut ...]
37.337.707(00.000.030): +-------------------------------+
37.337.707(00.000.000): |  192  |  168  |    1  |  217  | (dest)
37.337.738(00.000.031): +-------------------------------+
37.337.738(00.000.000): ip4_output_if: call netif->output()
37.337.768(00.000.030): pbuf_add_header: old 0x20028ff8 new 0x20028fea (14)
37.337.768(00.000.000): ethernet_output: sending packet 0x20028fd8
37.337.799(00.000.031): [   74675] ETHTX itf=w0 len=1514 src=28:cd:c1:ff:bf:1d dst=00:1b:21:f0:77:ee subtype=0800
37.337.951(00.000.152):     IPv4 tlen=1500 id=12 flags=0 frag_offset=0 protocol=6 src=192.168.001.230 dst=192.168.001.217
37.337.982(00.000.031):     TCP src=80 dst=42582 seq=0000196f ack=4dacfa16 flags=5018
37.338.012(00.000.030):     0x0000: 001b21f0 77ee28cd c1ffbf1d 08004500 : 05dc000c 0000ff06 3100c0a8 01e6c0a8 : ..!.w.(.......E. ........1.......
[... cut ...]
37.338.958(00.000.122):     0x0120: 69707420 7372633d 22757469 6c732e6a : 73223e3c 2f736372 6970743e 0a202020 : ipt src="utils.j s"></script>.   
37.339.050(00.000.092):     0x0140: 203c7363 72697074 20737263 3d226a71 : 75657279 2d6d696e 2e6a7322 3e3c2f73 :  <script src="jq uery-min.js"></s
37.339.172(00.000.122):     0x0160: 63726970 743e0a0a 20202020 203c7363 : 72697074 20207479 70653d22 74657874 : cript>..     <sc ript  type="text
37.339.263(00.000.091):     0x0180: 2f6a6176 61736372 69707422 3e0a2020 : 20202020 20242864 6f63756d 656e7429 : /javascript">.        $(document)
(snip)
</code>
</details>

eightycc, 2025-02-03:
This performance bug is giving me a chance to exercise my `ramlog` module, so just for that reason it's a useful exercise. Also, I enjoy shooting bugs that have a number of asynchronously interacting parts.

Consider my entries in this issue to be a work in progress. For example, yesterday I was convinced that a bug in LWIP was causing the performance glitch during a 3-way TCP handshake detailed above. On reflection and some more study I now see that it is not an LWIP bug but instead it is a combination of an LWIP configuration issue and the way that CircuitPython code handles the process of accepting an incoming TCP socket connection.

eightycc, 2025-02-03:
LWIP maintains a 250 ms "fast" timer and a 500 ms "slow" timer. These values probably made sense 20+ years ago, but now cause pain by inducing timeouts that are expensive to recover from. Configuring these to 25 and 50 ms resolves the 3-way TCP handshake delay by reducing the delay processing the first data packet to no more than 25 ms. 

LWIP memory management is almost infinitely configurable. Of course this is both a blessing and a curse. I've been experimenting with different static allocations, but that approach leads to solutions that work well for my tests but may be terrible for other use cases. Maybe use all dynamic allocation by substituting tlsf for all of LWIP's static and dynamic allocators?

jepler, 2025-02-04:
if I recall correctly I did the lwip implementation before we had tlsf.  I think having it allocate via tlsf is an interesting idea and could be worth pursuing.

eightycc, 2025-02-04:
@jepler @bablokb  In combination with LWIP periodic timer change, using tlsf for all of LWIP's memory needs nails it. I'm now seeing the original performance report running in about 700 ms. This also appears to fix intermittent "unknown error" reports from networking components. It also reduces the RAM footprint for LWIP significantly. Since this is a change that affects all CYW43-based networking on the RP2 port, I'll be submitting a pull for discussion and testing shortly.

bablokb, 2025-02-15:
I think this can be closed as #10027 is merged now.

closed 2025-02-15 by jepler