adafruit/circuitpython #9842: ESP8684 WiFi failing with MemoryError

detlevo, 2024-11-28:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.1 on 2024-11-20; ESP8684-DevKitC-02-N4 with ESP32C2
```

### Code/REPL

```python
import wifi
```

### Behavior

```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MemoryError: Failed to allocate Wifi memory```

### Description

Importing the wifi module fails with a MemoryError. WiFi functionality is non-functional thereafter.

tannewt, 2024-12-02:
This isn't a high priority for us because the C2 is very limited.

If you want to look yourself, then I'd suggest tweaking the sdkconfig to limit code loaded into RAM to execute. There may be optimization code that can live on flash instead.

detlevo, 2024-12-04:
As I am not familiar with the CircuitPython code internals I would like to know, what I would have to change in a sdkconfig file to achieve that goal. If it is too complicated I can live with the reported issue for the moment as I was more interested in interfacing my '[eric-ide](https://eric-ide.python-projects.org)' to this board.

nhedler-cubeworks, 2025-01-27:
I'm happy to help with testing but do not currently have the ability to set up the build process on my machine.