adafruit/circuitpython #9852: Raspberry Pi Pico 2 digitalio.DigitalInOut problems

tomiahola, 2024-12-02:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.1 on 2024-11-20; Raspberry Pi Pico 2 with rp2350a
```

### Code/REPL

```python
import time
import board
import digitalio

pin = board.GP15
gpio = digitalio.DigitalInOut(pin)
gpio.direction = digitalio.Direction.INPUT
for _ in range(4):
    gpio.pull = digitalio.Pull.UP
    print("read 1:", gpio.value)
    time.sleep(1)
    gpio.pull = digitalio.Pull.DOWN
    print("read 2:", gpio.value)
    time.sleep(1)
gpio.deinit()
```

### Behavior

code.py output:
read 1: True
read 2: True
read 1: True
read 2: True
read 1: True
read 2: True
read 1: True
read 2: True

### Description

[image]
Same problem occurs with multiple different pins. Also tested with 3 separate Pico 2 boards.

jepler, 2024-12-02:
Based on what you describe, you are encountering an erratum in the RP2350 silicon. The RP2350 datasheet proposes a workaround, but we [elected not to implement it in CircuitPython](https://github.com/adafruit/circuitpython/issues/9541#issuecomment-2341792598) for the following reasons:

> Without adding additional API specifically to support the RP2350, there is no way to inform CircuitPython in various use cases, "well, yes, this is a weak input signal, so do the workaround". The only hint would be if you enabled the internal pulldown, but there are other cases as well. We do not right now want to make a special-case API for the RP2350.
> 
> So our conclusion is to not implement the workaround, at least for now, but simply to document the restrictions that make it unnecessary: don't expect the internal pulldown to work, use an external pulldown 8.2k or lower if you need a pulldown, or make sure your input signal can sink enough current to override the leakage current. This is defining the problem away.
> 
> So this now becomes an issue to address via documentation.
> 
> In addition, we will disable touchio on RP2350 for now, because implementing it via an external pull-up is non-standard compared with all the other implementations, and that will be a support issue. (I personally think we could live with a non-standard impl, but I can see the argument the other way as well.) This might be revisited later.

closed 2024-12-02 by jepler

dhalbert, 2024-12-02:
There is a detailed writeup of the problem here: https://docs.circuitpython.org/en/latest/shared-bindings/digitalio/index.html#digitalio.DigitalInOut. See the "Warning box" further down the page.

jepler, 2024-12-02:
Out of personal curiosity, I implemented the E9 workaround in pure CircuitPython code: https://gist.github.com/jepler/81eef2bab4bb130bd2034d95f9c22b1d and lightly tested it on a Feather RP2350 using an pin with nothing connected to it.

jepler, 2024-12-04:
@dhalbert points out that my workaround code probably doesn't 100% work in practice, and a different physical design should be used to avoid the E9 erratum. Specifically, interrupts, gc, or background tasks can run between when the input buffer is enabled and the GPIO value is sampled. During that time, the extra leakage current might pull the input back up to HIGH.