adafruit/circuitpython #9856: wifi trouble : pystack exhausted

stanelie, 2024-12-03:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.0 on 2024-10-28; Waveshare ESP32-S3-Zero with ESP32S3
```

### Code/REPL

```python
import wifi
import time

def connect_to_wifi():
    try:
        print(f"connecting wifi")
        wifi.radio.connect("my_ssid","my_password", timeout=5)
    except ConnectionError as e:
        print(f"{e} . Retrying...")
        time.sleep(1)
        connect_to_wifi()
        
connect_to_wifi()
```

### Behavior

When trying to connect to a nonexistent access point, the ESP32 device will eventually crash after a minute or two with the following error message : 
`RuntimeError: pystack exhausted`

I would like the ESP32 to keep trying until it can connect, and not crash.

anecdata, 2024-12-03:
you could do `while not wifi.radio.connected:` inside the function, and add some time limit or other exit trigger to the conditional  if desired

stanelie, 2024-12-04:
Thanks for the tip, I will try that! Cheers!

stanelie, 2024-12-04:
Yup, that works great! Thanks again!

closed 2024-12-04 by stanelie