adafruit/circuitpython #9859: wrong temperature on RP2040 (it shows -1479.79)

aguaviva, 2024-12-04:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.1 on 2024-11-20; VCC-GND Studio YD RP2040 with rp2040
```

### Code/REPL

```python
import microcontroller
print(microcontroller.cpu.temperature)
```

### Behavior

it prints: `-1479.79`

eightycc, 2024-12-04:
Please see issue #8403. The Studio YD board has an ADC_VREF solder jumper that needs to be closed for CPU temperature to function.

jepler, 2024-12-04:
Thanks for linking to the solution to this @eightycc !

aguaviva, 2024-12-04:
It looks like the same is true to control the neopixel, there is a jumper one needs to solder. Thanks for sharing, I would have never realized it.