adafruit/circuitpython #9880: `display.root_group.hidden = True` causes `RuntimeError: Read-only`

dhalbert, 2024-12-08:
`display.root_group.hidden = True`
has error:

`RuntimeError: Read-only`

_Originally posted by @bh1rio in https://github.com/adafruit/circuitpython/issues/2791#issuecomment-2525231538_

dhalbert, 2024-12-08:
This started happening between 9.0.0-beta.0 and 9.0.0-beta.1.

dhalbert, 2024-12-08:
This is due to #8923.

@jepler 

`display.root_group.hidden = True` was previously recommended as a way to hide the terminal. Now that the root group is read-only, it can no longer be made hidden. I can't remember if there is an alternate way to hid the terminal now.

jepler, 2024-12-08:
I think it's `display.root_group = None`

dhalbert, 2024-12-08:
@jepler That makes sense, thanks! @bh1rio that should work for you. Please reply if not.

closed 2024-12-08 by dhalbert

dhalbert, 2024-12-08:
More on this: https://forums.adafruit.com/viewtopic.php?p=1019617#p1019617

@tannewt see your comments there.

bh1rio, 2024-12-08:
> More on this: https://forums.adafruit.com/viewtopic.php?p=1019617#p1019617
> 
> @tannewt see your comments there.

en...... set root_group to none, does not have effect

jepler, 2024-12-08:
Huh, I tested it here:
```
Adafruit CircuitPython 9.2.1-21-g7c28ec69cd-dirty on 2024-12-04; Adafruit Floppsy RP2040 with rp2040
>>> import board
>>> board.DISPLAY.root_group = None
```
After I hit enter on the 2nd line the TFT was set to all black.

I see two things from the forum thread:
 * first, you're using an epaper display ("pimoroni badger2040 which has a built in Epaper Display"). circuitpython will defer the refresh of an epaper display for a long time, sometimes minutes, based on the datasheet recommended refresh time/rate
 * second, based on the code sample you posted, your `code.py` finishes right after setting the root group to None (& calling refresh). However, whenever a program exits, the display is set back to the default, i.e., the console. You have to do something like have a loop that runs forever or enter a deep sleep mode to prevent this happening.

bh1rio, 2024-12-08:
it have effect in a vm, e.g. `code.py`.
after it run, display recovery.

if you use thonny, thonny will run `help('modules')`, and get dir list.
For the 320 * 240 LCD of M5Stack, it will take four to five screens to display all of these contents.

Perhaps it would be better for me to compile a bin with `CIRCUITPY_TERMINALIO=0` by myself.

thanks.

bh1rio, 2024-12-08:
`CIRCUITPY_TERMINALIO=0` is not a perfect solution either. By doing so, both the built-in `terminalio` and `fontio` modules will be gone, and there may be other modules that I haven't noticed are missing as well.

dhalbert, 2024-12-08:
You can try putting `board.DISPLAY.root_group = None`.

dhalbert, 2024-12-08:
You can try putting `board.DISPLAY.root_group = None` in `boot.py`. In my testing, this produced nothing on the screen when ctrl-C'ing a program, or when using Thonny.

todbot, 2025-02-22:
And for future people who discover this thread: you must put `board.DISPLAY.root_group = None` in both `boot.py` and `code.py` as it seems the display gets recreated after boot.py finishes. (at least on an M5Stack Cardputer)

konsumer, 2025-03-02:
`board.DISPLAY` is not defined on RP2040-zero with ssd1306. Is there another way to do this?

jepler, 2025-03-02:
`board.DISPLAY` is a short hand for "the built in display, if this board has one in its board definition". If you create a display object in your code with a line like `display = ssd1306(...)` then you'd write `display` instead of `board.DISPLAY`.

konsumer, 2025-03-03:
> board.DISPLAY is a short hand for "the built in display, if this board has one in its board definition". If you create a display object in your code with a line like display = ssd1306(...) then you'd write display instead of board.DISPLAY.

yep. If you do `display.root_group = None` with your own display, it doesn't work like that, though, is my point. At least on my RP2040, I needed to also run those 2 supervisor lines to remove the initial flash of console. Even with that, if your loop breaks, it will show the console, regardless of what you do with `display`.

juergenpabel, 2025-10-22:
Just for posterity: Hiding the REPL display output is still possible by hiding the REPL (UI-)elements themselves:
```
for item in displayio.CIRCUITPYTHON_TERMINAL:
    item.hidden = True
```
The contained elements are the CP logo, the status line plus the actual REPL content. Put this in boot.py and it'll be a blank screen (except for a short flicker before this code is actually executed).