adafruit/circuitpython #9886: Is max-width needed in the serial console?

FrozenGalaxy, 2024-12-12:
Is it really necessary to limit the width? Having it expand to actual width seems way more logical

[image]

Line in question: https://github.com/adafruit/circuitpython/blob/27456d525e645acc68126a3d4ad831aa01aa2281/supervisor/shared/web_workflow/static/style.css#L2

tannewt, 2024-12-12:
@andywarburton added it. They may have an opinion.

ThomasAtBBTF, 2024-12-12:
I would instead / or additionaly like to see a "maxcol" and "maxline" setting/style.
Lately, I have been working on a project with an "Adafruit Qualia ESP32-S3 for TTL RGB-666 Displays" board.
And "struggle" with the serial output of the different LCD panels.
"Somewhere" is the information about the currently used font and also the information about the currently connected LCD panel.
Bringing this together would be helpful!

andywarburton, 2024-12-12:
> @andywarburton added it. They may have an opinion.

Aye it was a was a while ago but generally super long lines are considered bad for readability (see https://baymard.com/blog/line-length-readability for example).

I’m not especially tied to the decision though so feel free to revert if folks think something else is better.

FrozenGalaxy, 2024-12-12:
> generally super long lines are considered bad for readability 

Well, the messages are long anyway, except its scroll able right now. 
It doesn't wrap lines/break the lines. 

So removing the max width has no downside, other than having it be able to display longer lines if your screen width is wider..
Your point would only make sense if the console had wrap lines enabled, which it doesn't.

[image]

RetiredWizard, 2024-12-14:
>And "struggle" with the serial output to the different LCD panels. "Somewhere" is the information about the currently used font and also the information about the currently connected LCD panel. Bringing this together would be helpful!

I don't know if this addresses the issues you're having, but this code has helped me deal with different display dimenstions:
```py
import board
from displayio import CIRCUITPYTHON_TERMINAL
from terminalio import FONT

height = round(board.DISPLAY.height/(FONT.bitmap.height*CIRCUITPYTHON_TERMINAL.scale))-1
width = round(board.DISPLAY.width/((FONT.bitmap.width/95)*CIRCUITPYTHON_TERMINAL.scale))-2
```

closed 2026-11-02 by dhalbert