adafruit/circuitpython #9891: MatrixPortal S3: Crash into Safemode after initalizing RGB matrix

RetiredWizard, 2024-12-16:
### CircuitPython version

```python
S3 Bucket: 2024-12-06T15:01:20.000Z adafruit-circuitpython-adafruit_matrixportal_s3-en_US-20241206-main-PR9873-c41d452.uf2:
Adafruit CircuitPython 9.2.1-26-gc41d4527b1 on 2024-12-06; Adafruit MatrixPortal S3 with ESP32S3

Artifact from #9889:
Adafruit CircuitPython 9.2.1-33-gbb49f770a1 on 2024-12-15; Adafruit MatrixPortal S3 with ESP32S3
```

### Code/REPL

```python
import board, displayio, framebufferio, rgbmatrix

displayio.release_displays()

addrPins = [
        board.MTX_ADDRA,
        board.MTX_ADDRB,
        board.MTX_ADDRC,
        board.MTX_ADDRD
]

matrix = rgbmatrix.RGBMatrix(
    width=64,height=32, bit_depth=2,
    rgb_pins=[
        board.MTX_R1,
        board.MTX_G1,
        board.MTX_B1,
        board.MTX_R2,
        board.MTX_G2,
        board.MTX_B2
    ],
    addr_pins=addrPins,
    clock_pin=board.MTX_CLK,
    latch_pin=board.MTX_LAT,
    output_enable_pin=board.MTX_OE,
    tile=1, serpentine=True,
)
```

### Behavior

```
Adafruit CircuitPython 9.2.1-33-gbb49f770a1 on 2024-12-15; Adafruit MatrixPortal S3 with ESP32S3
>>> import qmatrix
>>> 
>>> a

[12:35:00.454] Disconnected
[12:35:01.455] Warning: Could not open tty device (No such file or directory)
[12:35:01.455] Waiting for tty device..
[12:35:06.462] Connected
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

I actually didn't run the above code in code.py/main.py but manually imported from the REPL. I haven't tested yet what happens if the code is run from the REPL

After importing a file with the above code, if you type a random character into the REPL, which would normally give a NameError, the board crashes into safe mode.

jepler, 2024-12-16:
Is this new since 9.2.1? If so it could be https://github.com/adafruit/circuitpython/pull/9865 which was supposed to fix a problem with an incorrect initialization leading to a crash, but maybe I got it wrong.

jepler, 2024-12-16:
Thanks for testing that. We may just want to revert that PR in its entirety, since it only caused problems when a mistake was made in matrix initialization, rather than breaking them entirely.

RetiredWizard, 2024-12-24:
I think this has been resolved and Dan reopened the issue that prompted the now reverted update so this issue can probably be closed.

closed 2024-12-24 by RetiredWizard