adafruit/circuitpython #9893: Ability to create more than 1 MIDI `PortIn` or `PortOut`

probonopd, 2024-12-16:
Some MIDI controllers expose more than one MIDI interface, and I need to recreate that using Adafruit_CircuitPython_MIDI.

https://github.com/adafruit/circuitpython/blob/79fd8b7efcbceaa87456a971af8e1692e58807c2/shared-bindings/usb_midi/PortOut.c#L21-L24 says

> //|         """You cannot create an instance of `usb_midi.PortOut`.
//|
//|         PortOut objects are constructed for every corresponding entry in the USB
//|         descriptor and added to the ``usb_midi.ports`` tuple."""

It would be extremely useful if it was possible to create more than one MIDI interface per direction.

Reference:
* https://github.com/probonopd/MiniDexed/pull/743#issuecomment-2547049742

dhalbert, 2024-12-16:
The current MIDI descriptor we use is here: https://github.com/adafruit/circuitpython/blob/main/shared-module/usb_midi/__init__.c

To construct this dynamically would be a lot of code, and would not fit on some smaller boards that currently provide MIDI. An alternative would be to let the user supply their own descriptor, or some part of it. We allow that for custom HID devices.

If you could record and supply the fancier USB MIDI descriptor from an Arturia device, that would be helpful. Having said that, this is the first time we've gotten a request for fancier MIDI, so this would be a long-term project. We would welcome a generalized PR if anyone is interested in working on this.

probonopd, 2024-12-17:
Thanks @dhalbert. All I am looking for is to have 2 MIDI PortIn and 2 PortOut instead of 1 each, on the Pi Pico.

Currently I have:

```
midi = adafruit_midi.MIDI(midi_in=usb_midi.ports[0], midi_out=usb_midi.ports[1], in_channel=0, out_channel=0, in_buf_size=512, debug=debugging_on)
```

I would like to add:

```
midi1 = adafruit_midi.MIDI(midi_in=usb_midi.ports[2], midi_out=usb_midi.ports[3], in_channel=0, out_channel=0, in_buf_size=512, debug=debugging_on)
```

So are you saying that this would currently require rebuilding CircuitPython in C?

probonopd, 2024-12-18:
> If that is not possible, then being able to choose the MIDI "cable number" would already be helpful.

Looks like all I'd need is a way, from Python, to change the `0` to `1` there:

https://github.com/adafruit/circuitpython/blob/2d2fa28b4f290cba9ba403d8608cfadf043a0441/shared-module/usb_midi/PortOut.c#L13

dhalbert, 2024-12-18:
The current USB MIDI descriptor is fixed, and describes one MIDI device with two ports, one In and one Out. Do you want one device with four ports, two In and two Out? Or do you want two independent MIDI devices?

probonopd, 2024-12-18:
I am looking for either
* One device with dour ports, two In and two Out (I assume this makes the second set of ports use MIDI cable 1 rather than 0), or (if this is easier)
* A way to send and receive on MIDI cable 1 rather than 0