adafruit/circuitpython #9897: Support PIO pins >31

dhalbert, 2024-12-18:
rp2pio does not yet support GPIO numbers >31 and has some RP2040-specific includes. Generalize it for RP2350.

jepler, 2024-12-18:
I fear this turns out to be a bit wrinkly. I have a jumble of thoughts:

- GPIO base is per PIO, not per SM. So, we have a new problem when allocating SMs, that we have to know whether we want the high or low bank and whether a PIO has any other SMs associated with it. (pico-sdk solves this by requiring the PIO base register be set before calling pio_sm_set_config and erroring if it clearly doesn't match)
- There's an inherent ambiguity: On rp2040 & rp2350A, having a pin base of 20 and a count of 20 addresses pins 20..31 & 0..7 (wrap around arithmetic). on 2350b, should this address 20..39 or the same pins as on 2040?
  - on 2350b, the answer is that only the base pin is checked an so the effect depends on the pio base register setting
  - but if we want to make the bank selection implicit, we have to choose whether we want to favor compatible wrap-around (unlikely) or high bank use (likely)
- we check the `wait gpio` instruction for matching an otherwise used pin number, but this also depends on the bank setting.

dhalbert, 2024-12-18:
> * here's an inherent ambiguity: On rp2040 & rp2350A, having a pin base of 20 and a count of 20 addresses pins 20..31 & 0..7 (wrap around arithmetic). on 2350b, should this address 20..39 or the same pins as on 2040?

`GPIOBASE` can only be 0 or 16. From the datasheet description on `GPIOBASE`

> Relocate GPIO 0 (from PIO’s point of view) in the system GPIO numbering, to access more than 32 GPIOs from PIO.
> Only the values 0 and 16 are supported (only bit 4 is writable).

closed 2025-01-09 by tannewt