adafruit/circuitpython #9907: circuitpython for M5Stack Timer Camera X with ESP32 does not work with camera

simsong, 2024-12-22:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.1 on 2024-11-20; M5Stack Timer Camera X with ESP32

Board ID:m5stack_timer_camera_x
```

### Code/REPL

```python
import busio
import board
import espcamera
i2c = busio.I2C(scl=board.SCL, sda=board.SDA)
data_pins=[board.D2,board.D3,board.D4,board.D5,board.D6,board.D7,board.D8,board.D9]
c = espcamera.Camera(data_pins=data_pins,pixel_clock_pin=board.XCLK,vsync_pin=board.VSYNC,href_pin=board.HREF,i2c=i2c,pixel_format=espcamera.PixelFormat.JPEG)
```
```

### Behavior

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
espidf.IDFError: Operation or feature not supported
```

It is not clear how to use the camera. I can find no sample code using CircuitPython.
Note that the solution at https://github.com/adafruit/circuitpython/issues/9155 no longer works, because the `board` module no longer supports the "full example" that allegedly works under 9.1.3:

```
>>> dir(board)
['__class__', '__name__', 'BAT_ADC', 'BAT_HOLD', 'BM8563_I2C', 'BM8563_SCL', 'BM8563_SDA', 'D', 'D2', 'D3', 'D4', 'D5', 'D6', 'D7', 'D8', 'D9', 'G13', 'G4', 'HREF', 'I2C', 'LED', 'PCLK', 'RESET', 'SCL', 'SDA', 'SSCB_I2C', 'SSCB_SCL', 'SSCB_SDA', 'VSYNC', 'XCLK', '__dict__', 'board_id']
>>> 
```

and while we are at it:
```
>>> dir(espcamera)
['__class__', '__name__', 'Camera', 'FrameSize', 'GainCeiling', 'GrabMode', 'PixelFormat', '__dict__']
>>> 
```

### Description

There seems to be no obvious way open the camera from CircuitPython.

ladyada, 2025-07-04:
iirc we only have espcamera support for ESP32-S3

dhalbert, 2025-07-04:
The #9155 example is for M5Stack Core S3, ESP32-S3, but not ESP32. @bill88t added the Timer Camera board in #7941. @bill88t: It sounds like you got the camera to work. What version was it on? Do you have any sample code? Maybe it's lying around in your board 😃

paluigi, 2025-07-05:
Looking to the [comment](https://github.com/adafruit/circuitpython/pull/7941#issuecomment-1537447877) in [#7941](https://github.com/adafruit/circuitpython/pull/7941), it seems the `espcamera` module was not tested, and this is what causing errors I got...but this other [comment](https://github.com/adafruit/circuitpython/issues/7981#issuecomment-1553479090) suggest otherwise (unless @bill88t refers to the fact that he got it working on Arduino, as stated in the first comment).

If the issue confirmed, I'd suggest to edit the [board page](https://circuitpython.org/board/m5stack_timer_camera_x/) on CircuitPython until it's fixed, as it list the camera capability but there is no way to make it work.

bill88t, 2025-07-05:
Haven't tested in a very good while but it **used** to work. And pretty well at that.
I cannot physically test as I'm unable to access that board anytime soon
but it should work unless CircuitPython changed something.

I don't use espcamera as is, but use software on top of it (in CircuitPython).
Here is my config:
```py
data_pins = D
pixel_clock_pin = PCLK
vsync_pin = VSYNC
href_pin = HREF
i2c = i2c0
external_clock_pin = XCLK
external_clock_frequency = 20000000
powerdown_pin = None
reset_pin = RESET
```
with i2c0 being:
`SSCB_SCL SSCB_SDA`

If you craft a espload command with these params it should all work.

bill88t, 2025-07-05:
I use stuff™️ that calls it. So yes, it is being used.
I just don't have a espcamera oneliner to give.

bill88t, 2025-07-06:
It used to work. But as I said, I'm just unable to access it right now to produce the command.

If you give me a month to get my home renovation done, then yes, I'll be able to access the board and bring you the full command.
But until then this is all I have to give you from my existing stuff. I did give you everything I can at this time.

There are the PR notes and code comments I made while figuring how to port the camera, which you can dig through if you want more old stuff.

If you think removing the camera tag is a worthwhile change, you're more than welcome to create a pull request to the circuitpython.org repo.

> I don't think you are under any obligation to support this board nor the camera feature, but if you claim you do...then you should be able to provide some guidance for others to use.

You are correct that I do not support this board actively. I merely tune in *when I can* to help as much as possible.
I'm merely a contributor, porting whatever I can with my limited skills.