adafruit/circuitpython #9908: time.monotonic() runs faster than realtime for Adafruit Metro M7 Express under CP 9.2.1

twrackers, 2024-12-22:
I discovered that when a Metro M4 and Metro M7 are running the same code which acts upon a fixed period of time passing (measured by calls to **time.monotonic()**), my newly purchased M7 claims the time period has completed in only ~85% of the real time required by the M4. Both are sending begin and end messages via serial to a Raspberry Pi 400 which is listening to the serial via another Python code which measures the elapsed times. The Metro M4, as well as a Trinket M0, Gemma M0, and Circuit Playground Express, all come within 1% of the expected time.

Full details can be found in my forum post at [https://forums.adafruit.com/viewtopic.php?p=1039811](https://forums.adafruit.com/viewtopic.php?p=1039811). The test procedure I used, as well as results, are posted there as well.

twrackers, 2024-12-23:
Don't know if this is a clue, but the actual runtime was closer to 83% of what was expected, which is very close to 5/6. Maybe some counter or prescaler is specified incorrectly.

joehopfield, 2025-01-06:
Not a bug (except that monotonic() returning a 32bit float should not exist and I bet most naive uses are incorrect)

The test is misusing time.monotonic(), essentially comparing it across machines, and against actual elapsed time.

from the test:
> stop_at = time.monotonic() + time_sec
> ...
> while time.monotonic() < stop_at:
>     pass

docs:
> ... Only use it to compare against other values from time.monotonic() during the same code run.

On my esp32-s3 qt-py a loop like that only takes about 8 microseconds, and your M7 is much faster -  you're exceeding your 22 *bits* of precision in monotonic's short-float value.

(IMHO time.monotonic should either throw an exception when this happens or should be deprecated. 32bit float is inherently broken for time.monotonic's very vague semantics)

twrackers, 2025-01-06:
So time.monotonic() returns a float which is NOT measured in elapsed seconds? Does time.monotonic_ns() return integer nanoseconds, even if that clock does not have nanosecond resolution?  I'm actually trying to pace a loop of code that's running at 10 or 20 Hz while it interacts with code on another Arduino or CircuitPython micro.

I can expect that using too much precision will make the timing not quite periodic, but if monotonic or monotonic_ns are really returning seconds or nanoseconds, the average rate should be about right as long as the master clock oscillator isn't way off.

So what WOULD be a valid way to write a timing loop that forces code to pace itself at a desired rate if I can't trust the monotonic clocks?

joehopfield, 2025-01-07:
The docs (https://docs.circuitpython.org/en/latest/shared-bindings/time/#time.monotonic) suggest time.monotonic_ms() for measuring elapsed time safely. (your test code.py would be easy to change).

docs for the monotonic* funcs are inconsistent (is value "always increasing" or "can never go backward"? both descriptions are used) and has at least one typo/error where #monotonic_ns mentions monotonic.

If the M7 has a bug, a monotonic_ns version of the test will make a clean test case.

(I too wish there were better high resolution time control - squashing to milliseconds on 100+mhz cpus feels restrictive)

joehopfield, 2025-01-07:
Comparing a change in monotonic() to an external clock would seem to violate the first paragraph of its doc, no?

anecdata, 2025-01-07:
There is a calibration here:
https://docs.circuitpython.org/en/latest/shared-bindings/rtc/index.html
But it's relatively small and may not be enough to compensate for this.

joehopfield, 2025-01-07:
Sorry, I think I believed the "always increasing value" in the monotonic() doc. It is monotonic (never decreasing), not strictly increasing. I read "always" as implying it needed to increment by a ns to always increase.
Looking at the code in time/__init__.c, I see you're correct.
Sorry again - code > docs. :-P

twrackers, 2025-01-07:
> This test is not comparing to an external clock, it's comparing to itself to last 600 seconds.
> While the starting point of monotonic() is arbitrary, it's still supposed to measure seconds.

Neradoc is correct.

Metro M7 (device under test): Hello, Pi400, I'm going to use my clock to mark the start and end of what I consider to be a 10 minute interval. Please use your clock and tell the user how long you measured that time period to be. Minor variations from 10 minutes are to be expected.

Pi400: Wow, M7, I clocked you at _way less_ than 600 seconds.

twrackers, 2025-01-07:
> There is a calibration here: https://docs.circuitpython.org/en/latest/shared-bindings/rtc/index.html#rtc.RTC.calibration But it's relatively small and may not be enough to compensate for this.

I figure if I'm within one or two percent of the expected rate, I'm doing well enough. It's not a precision application I have in mind, nominal rate will probably be 10 Hz.

twrackers, 2025-01-07:
From Neradoc again:

> [00:35:46.584] Pro Micro
> [00:45:46.608] 600

Using Arduino's Serial Monitor with timestamps:
20:19:42.487 -> Metro M7
20:28:17.790 -> 600

Difference: 515.303 sec, 85.9% of expected 600 sec

(My) M7 is definitely running faster than it believes.

jepler, 2025-01-07:
The Metro M7 does not have a 32kHz crystal, which means the timing source is less accurate. 

I eventually found this information on the accuracy of the internal 32kHz clock source at https://www.digikey.com/en/htmldatasheets/production/4171371/0/0/1/mimxrt1011dae5a section 4.2.4.2:
> In addition, if the clock monitor determines that the OSC32K is not present, then the source of the 32 K will automatically switch to a crude internal ring oscillator. The frequency range of this block is approximately 10–45 kHz. It highly depends on the process, voltage, and temperature.

The value you're seeing corresponds to about 38kHz, which is within the range specified by NXP.

The design of the Metro M7 doesn't have a spot for a 32kHz crystal to be added, and CircuitPython's code doesn't currently allow the RTC to be based on the (more accurate) 24MHz crystal that is used to derive the CPU clock.

twrackers, 2025-01-07:
> The design of the Metro M7 doesn't have a spot for a 32kHz crystal to be added, and CircuitPython's code doesn't currently allow the RTC to be based on the (more accurate) 24MHz crystal that is used to derive the CPU clock.

Well pooh. :(

Guess that M7 will be relegated to tasks that don't require a really accurate timebase. Still got an M4 sitting next to it that does have more precise timing, it actually clocked in at 600.0 seconds on my timing test.

I suppose I could connect an external RTC to the M7 since I'm not looking at very high rate tasks. So far I see as options: 

- SparkFun RV-8803 [https://www.sparkfun.com/products/16281](https://www.sparkfun.com/products/16281)
- Adafruit DS3231 [https://www.adafruit.com/product/3013](https://www.adafruit.com/product/3013)

Both have battery backup and QWIIC connectors. The SparkFun part claims to read down to hundredths of a second while the Adafruit one reads down to seconds. And I just noticed I have one of the DS3231 parts sitting on my desk. Maybe I can code the M7 to self-calibrate its onboard RTC, say once every 10 seconds, so it can compensate for the ring oscillator on the fly. Guess I know what I'll be tinkering with tomorrow. ;)

twrackers, 2025-01-07:
> **jepler** changed the title ~~time.monotonic() runs faster than realtime for Adafruit Metro M7 Express under CP 9.2.1~~ Metro M7: without 32768Hz RTC crystal, RTC timekeeping is not accurate

Thanks, much better title for this thread.

dhalbert, 2025-01-07:
> CircuitPython's code doesn't currently allow the RTC to be based on the (more accurate) 24MHz crystal that is used to derive the CPU clock

@jepler Did you look in detail at this? Could we change the clock source?

twrackers, 2025-01-07:
> > CircuitPython's code doesn't currently allow the RTC to be based on the (more accurate) 24MHz crystal that is used to derive the CPU clock
> 
> @jepler Did you look in detail at this? Could we change the clock source?

That would be cool if it could be done... but I'm still going to spend part of today poking at using my DS3231 to clock the current timing live. I've already seen my test code give fairly different measures on the M7 on different days, probably because of variations in the room temperature. (Ooohh, that cries out for a speed-to-temperature calibration.)

But in the long run, changing the clock source would save some headache for future M7 users.

jepler, 2025-01-07:
I don't think there's an easy way to just make the existing RTC be clocked based on (division of) the 24MHz crystal.

from what I can tell, the RTC peripheral can only be clocked from the 32kHz crystal or the internal ring oscillator. there might be some other peripheral that can be used e.g., as a cycle counter but I don't know the imxrt series in that much detail.

So whatever the solution, for example with the Programmable Interrupt Timer (PIT), it would require changing the whole implementation of ticks in this port.

jepler, 2025-01-07:
In micropython, ports/mimxrt/ticks.c uses the GPT (general purpose timer) peripheral clocked from the 24MHz crystal (divided by 24, to give 1MHz) as the main timing source. This could be studied & adapted into CircuitPython, possibly.

twrackers, 2025-01-07:
Another Metro M7 oddity:

I'm beginning to think the Metro M7 should have been blue instead of red; I think it's actually a disguised TARDIS (wibbly-wobbly timey-wimey thing).

I connected a DS3231 RTC to the M7's QWIIC port, and ran the following test code.

```
import adafruit_ds3231
import time
import board

i2c = board.I2C()  # uses board.SCL and board.SDA
rtc = adafruit_ds3231.DS3231(i2c)

while True:
    t = rtc.datetime
    print(t)
    print(f'{t.tm_hour}:{t.tm_min}:{t.tm_sec}')
    time.sleep(1)
```

Even the time.sleep(1) is running faster than normal, about every seventh pair of lines printed replicates the pair of lines before. _(Note: struct_time(...) lines may appear truncated.)_

```
struct_time(tm_year=2025, tm_mon=1, tm_mday=7, tm_hour=18, tm_min=9, tm_sec=9, tm_wday=1, tm_yday=-1, tm_isdst=-1)
18:9:9
struct_time(tm_year=2025, tm_mon=1, tm_mday=7, tm_hour=18, tm_min=9, tm_sec=10, tm_wday=1, tm_yday=-1, tm_isdst=-1)
18:9:10
struct_time(tm_year=2025, tm_mon=1, tm_mday=7, tm_hour=18, tm_min=9, tm_sec=11, tm_wday=1, tm_yday=-1, tm_isdst=-1)
18:9:11
struct_time(tm_year=2025, tm_mon=1, tm_mday=7, tm_hour=18, tm_min=9, tm_sec=11, tm_wday=1, tm_yday=-1, tm_isdst=-1)
18:9:11
struct_time(tm_year=2025, tm_mon=1, tm_mday=7, tm_hour=18, tm_min=9, tm_sec=12, tm_wday=1, tm_yday=-1, tm_isdst=-1)
18:9:12
struct_time(tm_year=2025, tm_mon=1, tm_mday=7, tm_hour=18, tm_min=9, tm_sec=13, tm_wday=1, tm_yday=-1, tm_isdst=-1)
18:9:13
struct_time(tm_year=2025, tm_mon=1, tm_mday=7, tm_hour=18, tm_min=9, tm_sec=14, tm_wday=1, tm_yday=-1, tm_isdst=-1)
18:9:14
struct_time(tm_year=2025, tm_mon=1, tm_mday=7, tm_hour=18, tm_min=9, tm_sec=15, tm_wday=1, tm_yday=-1, tm_isdst=-1)
18:9:15
struct_time(tm_year=2025, tm_mon=1, tm_mday=7, tm_hour=18, tm_min=9, tm_sec=16, tm_wday=1, tm_yday=-1, tm_isdst=-1)
18:9:16
struct_time(tm_year=2025, tm_mon=1, tm_mday=7, tm_hour=18, tm_min=9, tm_sec=17, tm_wday=1, tm_yday=-1, tm_isdst=-1)
18:9:17
struct_time(tm_year=2025, tm_mon=1, tm_mday=7, tm_hour=18, tm_min=9, tm_sec=17, tm_wday=1, tm_yday=-1, tm_isdst=-1)
18:9:17
struct_time(tm_year=2025, tm_mon=1, tm_mday=7, tm_hour=18, tm_min=9, tm_sec=18, tm_wday=1, tm_yday=-1, tm_isdst=-1)
18:9:18
```

twrackers, 2025-01-07:
Forgot to mention: the DS3231 has a battery installed and was preset to GMT so it's five hours ahead of my local time (US Eastern Standard Time). Not that it really matters right now.

jepler, 2025-01-07:
sleep is based on the same internal time source as monotonic time.