adafruit/circuitpython #9914: Makerfab 7tft has no CS for sdcard operation

mwalimu, 2024-12-25:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.1 on 2024-11-20; MakerFabs-ESP32-S3-Parallel-TFT-With-Touch-7inch with ESP32S3
Board ID:makerfabs_tft7
UID:48CF6E86FA45
```

### Code/REPL

```python
>>> import board
>>> dir(board)
['__class__', '__name__', 'BUTTON', 'DISPLAY', 'GPIO19', 'GPIO20', 'I2C', 'RX', 'SCL', 'SDA', 'SDIO_CLK', 'SDIO_CMD', 'SDIO_D0', 'TFT_BACKLIGHT', 'TFT_PINS', 'TFT_TIMINGS', 'TFT_TIMINGS1024', 'TFT_TIMINGS800', 'TOUCH_RESET', 'TX', '__dict__', 'board_id']
>>>
```

### Behavior

Can't use SD card since there is no chip select defined in "board".

snkYmkrct, 2024-12-26:
@mwalimu 
The interface used for the SD card on this board is SDIO, not SPI. 
It's not clear from the schematic, and the documentation wiki is broken, but there are comments in their [sdcard code example.](https://github.com/Makerfabs/ESP32-S3-Parallel-TFT-with-Touch-7inch/blob/13f5bce5b168a5feb66619f433d4058977ec82f2/example/sdio_demo/sdio_demo.ino#L15)

The `sdioio` module should be used instead of `sdcardio` , see:  https://docs.circuitpython.org/en/latest/shared-bindings/sdioio/index.html and https://learn.adafruit.com/adafruit-microsd-spi-sdio/using-sdioio

RetiredWizard, 2024-12-26:
I loaded this on the Makerfab 7tft to mount an SD card:
```py
import os, storage, board, sdioio
sd =sdioio.SDCard(clock=board.SDIO_CLK,command=board.SDIO_CMD,data=[board.SDIO_D0],frequency=25000000)
vfs = storage.VfsFat(sd)
storage.mount(vfs,'/sd')
os.listdir('/sd')
```

dhalbert, 2024-12-29:
@mwalimu Does this solve your issue?

mwalimu, 2024-12-29:
Yes it did.  While it is true that there is no CS pin, the error was that I forgot to make board.SDIO_D0 a tupple.

DUH!  (face plant)

closed 2025-01-02 by tannewt