adafruit/circuitpython #9915: Circuitpython 9.2.1 fails to run on EPS32P4-Function-EV board

ralphrmartin, 2024-12-26:
### CircuitPython version

```python
9.2.1 for ESP32 P4
Also tried later development versions with the same outcome.
```

### Code/REPL

```python
Didn't get a repl
```

### Behavior

After successfully flashing circuitpython 9.2.1 (for esp32 p4) using esptool, I get no repl, and the serial monitor just shows 

```
SP-ROM:esp32p4-eco2-20240710
Build:Jul 10 2024
rst:0x10 (CHIP_LP_WDT_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x78
load:0x4ff2abd0,len:0x860
load:0x4ff2cbd0,len:0x23a0
entry 0x4ff2abd0

```
repeatedly.

### Description

Using ESP32 P4 Function EV board with connected camera and lcd.

### Additional information

I also cannot get a repl using thonny.

ralphrmartin, 2024-12-26:
I have successfully run esp's own examples for the board, so the hardware is OK.

dhalbert, 2024-12-26:
Did you erase the flash before loading the .bin?

Try 9.2.0, and also try the latest version, both from here: https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/espressif_esp32p4_function_ev/

ralphrmartin, 2024-12-26:
Yes, I erased the flash first. I get the same result with 9.2.0.

In Thonny it reports 
```
Device is busy or does not respond. Your options:

  - wait until it completes current work;
  - use Ctrl+C to interrupt current work;
  - reset the device and try again;
  - check connection properties;
  - make sure the device has suitable MicroPython / CircuitPython / firmware;
  - make sure the device is not in bootloader mode.

```
but I get the same after pushing RST.

ralphrmartin, 2024-12-26:
The latest version does not work either.

ralphrmartin, 2024-12-26:
Further info - the web installer does not recognise the chip, 
```
Detecting chip type... unknown!
Error: Cannot read properties of null (reading 'getChipDescription')

```so I used e.g.
`esptool.py --port /dev/tty.usbserial-210 write_flash -z 0x0 /Users/ralph/Downloads/adafruit-circuitpython-espressif_esp32p4_function_ev-en_US-9.2.0.bin 
`

dhalbert, 2024-12-26:
@tannewt has this board and did the port. I don't, unfortunately, so I think we'll have to wait for his return from break in early January.

RetiredWizard, 2024-12-27:
I loaded 9.2.1 by connecting to the type C connector labeled: USB-UART and running this command:
```
python ~/micros/bin/esptool.py --chip esp32p4 --port /dev/ttyUSB0 write_flash -z 0x0000 ~/Downloads/adafruit-circuitpython-espressif_esp32p4_function_ev-en_US-9.2.1.bin
```
After the firmware was flashed I was able to connect to the REPL using tio on the /dev/ttyUSB0 port. If after flashing the firmware I connected to the type C connector labeled USB-OTG no serial port was created and of course the CIRCUITPY drive didn't show up presumably because that PR hasn't been merged in to "stable" yet.

I also loaded the "Absolute Newest" via the USB-UART type C connector using:
```
python ~/micros/bin/esptool.py --chip esp32p4 --port /dev/ttyUSB0 write_flash -z 0x0000 ~/Downloads/adafruit-circuitpython-espressif_esp32p4_function_ev-en_US-20241222-main-PR9906-53f76c5.bin
```
After which, the REPL was not available on the created /dev/ttyUSB0 port, but if I plugged into the type C connector labeled USB-OTG the port /dev/ttyACM0 was created and provided REPL access. Also the CIRCUITPY drive was mounted.

ralphrmartin, 2024-12-27:
Before starting, I had the P4 on /dev/tty.usbserial-210 when plugged into the USB-UART.
I downloaded using
`esptool.py --chip esp32p4 --port /dev/tty.usbserial-210 write_flash -z 0x0000 /Users/ralph/Downloads/adafruit-circuitpython-espressif_esp32p4_function_ev-en_GB-9.2.1.bin`
After downloading, I checked for other serial ports, but I could only still see /dev/tty.usbserial-210, with no REPL. I can see no tty on connecting to the USB-OTG port. Doing 
`screen /dev/tty.usbserial-210 115200`
gives no REPL, just the output reported above.

After again erasing flash, I tried doing the same again, but with the latest version
`esptool.py --chip esp32p4 --port /dev/tty.usbserial-210 write_flash -z 0x0000 /Users/ralph/Downloads/adafruit-circuitpython-espressif_esp32p4_function_ev-en_US-20241222-main-PR9906-53f76c5.bin
with the same result: no tty on the USB-OTG port, and no REPL on /dev/tty.usbserial-210

This is all under MacOS 15.2, in case that matters.
`

ralphrmartin, 2024-12-27:
Furthermore, I am using esptool from esp-idf 5.5, in case that is relevant.

ralphrmartin, 2024-12-27:
My ESP-32 P4 board is version v1.4

ralphrmartin, 2024-12-27:
I tried disconnecting the display, but it made no difference. (I didn't want to disconnect the camera is I dont want to risk mangling the fragile edge connector).

dhalbert, 2024-12-27:
So _something_ is different. It sounds like you are using the same ports, but here's a picture -- maybe you can both describe exactly which ports you are using:
(from https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32p4/esp32-p4-function-ev-board/user_guide.html)
[image]

What is the version number printed on the board?
What is the part number of the ESP32-P4?
What is the part number of the flash chip that's on the board?

I don't see any jumpers to change.

ralphrmartin, 2024-12-27:
When flashing, I used the USB-to-UART port, as stated above. I get no REPL on either USB-to-UART or USB 2.0 Type C (USB-OTG) port in either case. When applying additional power I also connected a PD charger to the USB Power in Port.
The LCD display is connected to the board's 5V ans GND, but this should OK with the extra PD charger connect. I had no power issues when using the device without the extra PD charger when running Espressif's demos, even ones using the LCD and WiFi at the sme time.

My Board is V1.4, as stated. My P4 chip says
```
ESP32-P4
NRW32
FC00
FEBJ012

```
My flash chip says
```
GigaDevice
25Q128ESIG
UMB127
AM2425

```

RetiredWizard, 2024-12-27:
Another interesting difference is that I don't see the serial output you're original post describes on either port using the 9.2.1 firmware but it does show up on the USB-to-UART port after flashing the "absolute newest" firmware image. With that firmware the REPL and CIRCUITPY drive show up over the port labeled "USB 2.0 Type-C Port".

dhalbert, 2024-12-27:
@ralphrmartin you are seeing a boot loop, assuming you see that output over and over again no matter what. If anything, your chip looks later based on the version or date codes, but I haven't been able to find a discussion of the codes.

ralphrmartin, 2024-12-27:
I am also using esptool v4.8.1. I agree that I am stuck in a boot loop (watchdog reset) and I agree that my chip looks later.

dhalbert, 2024-12-27:
Could you each do:
```
esptool.py --port ... chip_id
```
so we can see if the chips are different revs?

RetiredWizard, 2024-12-27:
```
esptool.py v4.8.1
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-P4 (revision v0.1)
Features: High-Performance MCU
Crystal is 40MHz
MAC: 60:55:f9:f9:88:2d
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-P4 has no Chip ID. Reading MAC instead.
MAC: 60:55:f9:f9:88:2d
Hard resetting via RTS pin...
```
I also connected the camera to my board and it made no difference....

ralphrmartin, 2024-12-27:
`esptool.py --port /dev/tty.usbserial-210 --chip esp32p4 read_mac`
reports a mac address which is not 00:00:00:00:00:00, so I conclude my part is not an engineering preproduction sample.

ralphrmartin, 2024-12-27:
I get 
```
esptool.py v4.8.1
Serial port /dev/tty.usbserial-210
Connecting....
Chip is ESP32-P4 (revision v1.0)
Features: High-Performance MCU
Crystal is 40MHz
MAC: 60:55:f9:fc:00:42
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-P4 has no Chip ID. Reading MAC instead.
MAC: 60:55:f9:fc:00:42
Hard resetting via RTS pin...

```

dhalbert, 2024-12-27:
OK, there we go!
`Chip is ESP32-P4 (revision v0.1)`
vs
`Chip is ESP32-P4 (revision v1.0)`
I wonder if we need to upgrade ESP-IDF to handle something about the newer revision.

ralphrmartin, 2024-12-27:
This page
`https://espressif.com.cn/sites/default/files/ae/ESP32-P4%20(Revision%20V1.0)%20Engineering%20Sample%20Notes_EN.pdf`
says

> Please note that before using the current ESP32-P4 Revision v1.0 chip, you must
> upgrade t o ESP-IDF master branch commit ID:
> 0546805fc57422a52ef390ad70a4bfcd92a42f6c or a later version.

dhalbert, 2024-12-27:
Thanks for finding that. I did some looking: those changes are in ESP-IDF v5.3.2 -- we are at v5.3.1, so an upgrade would fix that. I tried just substituting the changed files from that commit but there are other changes that make it not compile. I opened issue #9917.

ralphrmartin, 2024-12-27:
Great thanks. I'll wait patiently...

RetiredWizard, 2024-12-27:
I have no idea what's involved in updating the IDF version but just to see what would happen, I updated the version listed in .gitmodules and compiled a firmware image. The image loads on my board but that obviously doesn't mean the firmware is using 5.3.2. I'm curious if there's any chance this image is using 5.3.2 and if it will work with the v1.0 P4 board.

[firmware.zip](https://github.com/user-attachments/files/18263578/firmware.zip)

ralphrmartin, 2024-12-27:
Yes - I tried it out but it ended up the same way as the other failures....
Thanks anyway! :-)

closed 2025-01-09 by dhalbert

ralphrmartin, 2025-01-09:
Confirmed this is fixed, but I have to reconnect to the USB device port after downloading (as per Retired Wizard's description, which is rather inconvenient, but necessary I suppose to get the the USB mass storage device to show up.

PR #9928 (2025-01-09): Update to IDF 5.3.2
Fixes #9917. Fixes #9915
commit 62525e9e (2025-01-09): Update to IDF 5.3.2

Fixes #9917. Fixes #9915