adafruit/circuitpython #9920: Alarm module not included for rp2350?

moviebrain, 2024-12-28:
This is going to be one of those issues where if I knew more I might not have to ask about it. I noticed that the 9.2.1 firmware for Pico 2 (rp2350) currently doesn't include the Alarm module. I briefly looked at the shared-bindings code in the repo (and I am completely incompetent w/ C) but I didn't see any code in there that looked specific to implementing alarms per-board.

Can I ask about progress or progression blocks for this module specific to the rp2350? Was it an intentional removal? Is there a better way to poll for a button press to wake from deep sleep with the newer chip that I'm unaware of w/ circuitpython?

 Thanks in advance.

dhalbert, 2024-12-28:
Sleep works differently on RP2350 than RP2040. We just haven't gotten around to it yet.

Neradoc, 2024-12-28:
Hi, if you're curious about the project structure, the port-specific implementations are found in the `ports/<portname>/common-hal/` directories (and supporting libraries in the port directory). `shared-bindings` define the python interface to modules and `shared-module` common implementations.

Here is the rp2 implementation:
https://github.com/adafruit/circuitpython/tree/main/ports/raspberrypi/common-hal/alarm

And where it's disabled for 2350:
https://github.com/adafruit/circuitpython/blob/27beb0efbe1f22ad970b1640e9908213c52dd1c6/ports/raspberrypi/mpconfigport.mk#L58-L60

moviebrain, 2024-12-29:
I super appreciate the explanation, I clearly didn't check deep enough. Thanks for your time explaining, I wish this (or the SDK examples I've looked at) were within my grasp to help add. I appreciate all the hard work of the contributors.

tannewt, 2025-01-02:
This is a duplicate of https://github.com/adafruit/circuitpython/issues/9491

closed 2025-01-02 by tannewt