adafruit/circuitpython #9921: os file system time not synchronized

TinkerPhu, 2024-12-28:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.1 on 2024-11-20; Adafruit QT Py ESP32S2 with ESP32S2
```

### Code/REPL

```python
#===========boot.py:
import storage
storage.remount("/", readonly=False, disable_concurrent_write_protection=True)

#===========code.py:
import adafruit_connection_manager
import adafruit_ntp
import os
import time
import wifi
import rtc

wifi_ssid = os.getenv("CIRCUITPY_WIFI_SSID")
wifi_password = os.getenv("CIRCUITPY_WIFI_PASSWORD")
wifi.radio.connect(wifi_ssid, wifi_password)

pool = adafruit_connection_manager.get_radio_socketpool(wifi.radio)
ntp = adafruit_ntp.NTP(pool, tz_offset=1, cache_seconds=3600)
path = "/sd/hello.txt"

print(time.localtime())
print(ntp.datetime)
rtc.set_time_source(ntp)
print(time.localtime())

while True:
    print(time.localtime())
    
    with open(path, "w") as f:
        f.write("hello")

    stats = os.stat(path)
    file_size = stats[6]
    file_date = stats[9]
    print("file_date:",time.localtime(file_date))
    time.sleep(5)
```

### Behavior

file date is not synchronized with the system time.

code.py output:
struct_time(tm_year=2000, tm_mon=1, tm_mday=1, tm_hour=1, tm_min=0, tm_sec=14, tm_wday=5, tm_yday=1, tm_isdst=-1)
struct_time(tm_year=2024, tm_mon=12, tm_mday=28, tm_hour=21, tm_min=56, tm_sec=10, tm_wday=5, tm_yday=363, tm_isdst=-1)
struct_time(tm_year=2024, tm_mon=12, tm_mday=28, tm_hour=21, tm_min=56, tm_sec=10, tm_wday=5, tm_yday=363, tm_isdst=-1)
struct_time(tm_year=2024, tm_mon=12, tm_mday=28, tm_hour=21, tm_min=56, tm_sec=15, tm_wday=5, tm_yday=363, tm_isdst=-1)
**_file_date: struct_time(tm_year=2000,_** tm_mon=1, tm_mday=1, tm_hour=1, tm_min=0, tm_sec=18, tm_wday=5, tm_yday=1, tm_isdst=-1)
struct_time(tm_year=2024, tm_mon=12, tm_mday=28, tm_hour=21, tm_min=56, tm_sec=20, tm_wday=5, tm_yday=363, tm_isdst=-1)
file_date: struct_time(tm_year=2000, tm_mon=1, tm_mday=1, tm_hour=1, tm_min=0, tm_sec=24, tm_wday=5, tm_yday=1, tm_isdst=-1)

### Additional information

This bug does not happen on XIAO ESP32-C3 with the same code. On XIAO ESP32-C3 there seams to be an background worker that syncronizes with ntp as soon as connection to internet is made. It does not need a manual ntp request.

Neradoc, 2024-12-28:
Hi, I see the same behavior on both an S2 and C3 QT PY.

It seems that the file time does not follow `time.localtime()` (set by the time source) but the internal RTC time.
I don't know enough about that to say if it's a bug or a (undocumented ?) expected behavior.
(The C code for file access might not want to rely on calling a python time source).

But file times are correct for me when not setting the time source and instead use:
```py
rtc.RTC().datetime = ntp.datetime
```
Which might mean that you need to manually call that from time to time to keep it in sync.
Also note that the RTC time should be kept when the code is reloaded until the board is hard reset.

I see no difference between the QT PY C3 and the QT PY S2 with the same code (in both your code, and my modified version the boards behave identical to each other). Maybe you set the RTC on the C3 in a previous test or some other code ?

First run after changing the code as mentioned above:
```
struct_time(tm_year=2000, tm_mon=1, tm_mday=1, tm_hour=0, tm_min=0, tm_sec=24, tm_wday=5, tm_yday=1, tm_isdst=-1)
struct_time(tm_year=2024, tm_mon=12, tm_mday=28, tm_hour=23, tm_min=11, tm_sec=19, tm_wday=5, tm_yday=363, tm_isdst=-1)
struct_time(tm_year=2024, tm_mon=12, tm_mday=28, tm_hour=23, tm_min=11, tm_sec=19, tm_wday=5, tm_yday=363, tm_isdst=-1)
struct_time(tm_year=2024, tm_mon=12, tm_mday=28, tm_hour=23, tm_min=11, tm_sec=19, tm_wday=5, tm_yday=363, tm_isdst=-1)
file_date: struct_time(tm_year=2024, tm_mon=12, tm_mday=28, tm_hour=23, tm_min=11, tm_sec=18, tm_wday=5, tm_yday=363, tm_isdst=-1)
```
Second run (time kept after soft reload)
```
struct_time(tm_year=2024, tm_mon=12, tm_mday=28, tm_hour=23, tm_min=11, tm_sec=30, tm_wday=5, tm_yday=363, tm_isdst=-1)
struct_time(tm_year=2024, tm_mon=12, tm_mday=28, tm_hour=23, tm_min=11, tm_sec=31, tm_wday=5, tm_yday=363, tm_isdst=-1)
struct_time(tm_year=2024, tm_mon=12, tm_mday=28, tm_hour=23, tm_min=11, tm_sec=31, tm_wday=5, tm_yday=363, tm_isdst=-1)
struct_time(tm_year=2024, tm_mon=12, tm_mday=28, tm_hour=23, tm_min=11, tm_sec=31, tm_wday=5, tm_yday=363, tm_isdst=-1)
file_date: struct_time(tm_year=2024, tm_mon=12, tm_mday=28, tm_hour=23, tm_min=11, tm_sec=30, tm_wday=5, tm_yday=363, tm_isdst=-1)
```

TinkerPhu, 2025-01-12:
Thank you and sorry for returning so late!

`rtc.RTC().datetime = ntp.datetime
`
This solved my problem!

Regarding C3: mind that I talk about the XIAO ESP32-C3 module, not adafruits QT module. They XIAOs C3 and adafruits C3 behave different. But with this solution, I can make them behave the same. Thanks again!

closed 2025-01-12 by TinkerPhu