adafruit/circuitpython #9924: ESP32-S2 (LOLIN S2 Mini) does not show CircuitPython USB drive

probonopd, 2024-12-29:
### CircuitPython version

```python
https://circuitpython.org/board/lolin_s2_mini/ 9.2.1
```

### Code/REPL

```python
None
```

### Behavior

* Windows 11
* `esptool.exe --chip esp32s2   --baud 460800  write_flash -z 0x0 combined.bin`
* Push reset
* Blue LED comes up permanently, a drive D: shows up as "S2MINIBOOT"
* Copy `adafruit-circuitpython-lolin_s2_mini-en_US-9.2.1.uf2` to that drive. During the copy, the blue LED is flashing. After a while, the "device" removed sound plays and drive D: disappears
* I would expect that a "device attached" sound would play and a CircuitPython drive would appear, but that does not happen

### Additional information

```
TinyUF2 Bootloader 0.20.1 - tinyusb (heads/master)
Model: Lolin S2 Mini
Board-ID: ESP32S2FN4R2-S2Mini-1-0-0
Date: Sep  5 2024
Flash Size: 0x00160000 bytes
```

probonopd, 2024-12-29:
On https://circuitpython.org/board/lolin_s2_mini/ it says:

> Note: `update.uf2` files are not currently working on ESP32-S2 or ESP32-S3 boards.

Does this mean that copying `adafruit-circuitpython-lolin_s2_mini-en_US-9.2.1.uf2` as described will NOT work? What is the correct procedure then?

Flashing `adafruit-circuitpython-lolin_s2_mini-en_US-9.2.1.bin` as follows will also NOT make a CircuitPython drive show up:

`esptool.exe --chip esp32s2 --baud 460800  write_flash -z 0x0  adafruit-circuitpython-lolin_s2_mini-en_US-9.2.1.bin`

probonopd, 2024-12-29:
https://learn.adafruit.com/circuitpython-with-esp32-quick-start/overview says

> It is important to remember that the ESP32 lacks native USB support and therefore no CIRCUITPY folder will show up. Loading CircuitPython firmware and getting wifi set up is done over a serial connection (a.k.a a COM port) instead of a folder.

If this is true, then this fact should imho be displayed very prominently on https://circuitpython.org/board/lolin_s2_mini/.

But at least the ESP32-S2 is capable of USB device mode (in fact, the uf2 bootloader is doing exactly that!), so what is holding it back from showing up as a CircuitPython USB device?

probonopd, 2024-12-29:
When using Thonny, I get

```
Downloading from https://downloads.circuitpython.org/bin/lolin_s2_mini/en_US/adafruit-circuitpython-lolin_s2_mini-en_US-9.2.1.uf2
(..)
100%Copying to D:\adafruit-circuitpython-lolin_s2_mini-en_US-9.2.1.uf2
Writing to D:\adafruit-circuitpython-lolin_s2_mini-en_US-9.2.1.uf2
100%
Waiting for the port...
Warning: Could not find port in 10 seconds
```

So also via the serial port no connection seems to be possible, leaving no way to use CircuitPython on the device.

dhalbert, 2024-12-29:
> > Note: `update.uf2` files are not currently working on ESP32-S2 or ESP32-S3 boards.
> 
> Does this mean that copying `adafruit-circuitpython-lolin_s2_mini-en_US-9.2.1.uf2` as described will NOT work? What is the correct procedure then?

No, that remark is about the UF2 for updating the bootloader, not CircuitPython.

> https://learn.adafruit.com/circuitpython-with-esp32-quick-start/overview says
>
>
> If this is true, then this fact should imho be displayed very prominently on [...]

That remark is about the plain ESP32, which is a different chip than the ESP32-S2. The chip names are confusing.

dhalbert, 2024-12-29:
> I would expect that a "device attached" sound would play and a CircuitPython drive would appear, but that does not happen.

If you press reset or unplug and replug the device, does it then appear?

There may well be a bug. This board definition was contributed by a community member. It may also be that the specs of this board are different than the original. There are many variations and even sometimes counterfeits of these sorts of boards. Could you provide a link to the product page for what you bought?

probonopd, 2024-12-29:
Thanks @dhalbert for looking into this.

> If you press reset or unplug and replug the device, does it then appear?

No.

> There may well be a bug. This board definition was contributed by a community member. It may also be that the specs of this board are different than the original. 

Well. It works for the uf2 bootloader, but not for CircuitPython. So I think whatever hardware configuration the uf2 bootloader is using, CircuitPython should also use, and then it should work, no?

> There are many variations and even sometimes counterfeits of these sorts of boards. Could you provide a link to the product page for what you bought?

The board says "S2 Mini v1.0.0 WEMOS.CC" on the PCB. I got it from some Asian store a while ago. But again, since showing the USB drisk is working with the uf2 bootloader, I doubt it is a hardware issue.

Is there anything I can do to help debugging this?

bablokb, 2024-12-29:
I have a few of these boards. They are most probably counterfeits, unless you buy directly from the Wemos store.

Nevertheless, my boards work fine with CircuitPython. Just retested after flashing the 9.2.1 (German version). Not only does the CIRCUITPY drive show up, it also connects to  my WLAN. Getting into UF2 bootloader mode is a bit tricky though and I needed a few tries. I am not sure if Thonny does the right thing. I would try manually until you see a drive with about 32MB (compared to the 1MB of the CIRCUITPY drive).

probonopd, 2024-12-29:
@bablokb which exact file are you flashing with which exact command? Thank you very much.

probonopd, 2024-12-29:
Wow, after erasing and then uploading the BIN version for https://circuitpython.org/board/espressif_esp32s2_devkitc_1_n4/ it is working now. I now see the CIRCUITPY drive.
```
Adafruit CircuitPython 9.2.1 on 2024-11-20; ESP32-S2-DevKitC-1-N4 with ESP32S2
Board ID:espressif_esp32s2_devkitc_1_n4
```

What I would suggest is a "S2 Mini without PSRAM" variant to be added to the download page.

These boards are advertised honestly as "without PSRAM" by many sellers. So I would not call them "counterfeit", but a "low cost version".

[image]