adafruit/circuitpython #9931: ESP32- S2 Reverse TFT Feather - i2c scan returns multiple bogus addresses

T-Mosher, 2025-01-03:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.1 on 2024-11-20; Adafruit Feather ESP32-S2 Reverse TFT with ESP32S2
Board ID:adafruit_feather_esp32s2_reverse_tft
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2021 Kattni Rembor for Adafruit Industries
# SPDX-License-Identifier: MIT
"""CircuitPython I2C Device Address Scan"""
import time
import board

#i2c = board.I2C()  # uses board.SCL and board.SDA
# Note; it's the same interface, but turns on the STEMMA QT port power
i2c = board.STEMMA_I2C()  # For using the built-in STEMMA QT connector on a microcontroller

while not i2c.try_lock():
    pass

try:
    while True:
        print(
            "I2C addresses found:",
            [hex(device_address) for device_address in i2c.scan()],
        )
        time.sleep(2)

finally:  # unlock the i2c bus when ctrl-c'ing out of the loop
    i2c.unlock()
```

### Behavior

Testing just the ESP32-S2 Reverse TFT Feather (no additional Feather boards connected, nothing on the Stemma_QT port), the I2C scan returns many incorrect I2C addresses.

[image]

In this configuration, the only I2C address that should appear is for the built-in MAX17048 battery gauge on [0x36].

### Description

Does not matter whether the i2c_scan uses board.I2C() or board.STEMMA_I2C(). 
Does not matter whether a LiPo battery is connected to the Feather.
Does not matter whether any other Feather modules are stacked with the ESP32-S2 Feather.
All of these variants result in the same issue: bogus addresses being detected.

### Additional information

See this Forum thread for complete details (including investigation by "mikeysklar").
https://forums.adafruit.com/viewtopic.php?t=215740

The Forum thread has a lot of pictures showing a stack of Feathers attached to the ESP32-S2, but that doesn't matter. The issue happens even with no additional boards installed.

The investigation by mikeysklar shows that the issue happens on the ESP32-S2 Reverse TFT, but not on the ESP32-S3 Reverse TFT.

jposada202020, 2025-01-03:
The problem also appears in the feather TFT ESP32-S2
tested in Adafruit CircuitPython 9.2.1 on 2024-11-20; Adafruit Feather ESP32-S2 TFT with ESP32S2

```

>>> import ms
I2C addresses found: ['0xb', '0x2b', '0x36', '0x3f', '0x65']
I2C addresses found: ['0x9', '0xf', '0x18', '0x5f', '0x6b']
I2C addresses found: ['0x9', '0x12', '0x2a', '0x5f']
I2C addresses found: ['0x9', '0x14', '0x15', '0x5f', '0x6e', '0x77']
I2C addresses found: ['0x9', '0x14', '0x18', '0x5e', '0x66']
I2C addresses found: ['0x9', '0x16', '0x1b', '0x3a', '0x60', '0x73']
I2C addresses found: ['0x9', '0xe', '0x1f', '0x2b', '0x34', '0x61']

```

mikeysklar, 2025-01-03:
```
+---------+-----------------+
| Version | Result          |
+---------+-----------------+
| 9.2.1   | ghost addresses |
| 9.1.4   | no devices      |
| 9.1.3   | no devices      |
| 9.1.0   | no devices      |
| 9.0.5   | works           |
| 8.2.10  | works           |
+---------+-----------------+
```

dhalbert, 2025-01-08:
This is probably because of #9671, which switched to using the ESP-IDF routine `i2c_master_probe()` to probe for devices. Previously we were doing the probing "manually", and that caused #9535 and #9561. I will see if some adjustment can be made.

dhalbert, 2025-01-08:
I think I have a fix for this via #9942. If you want to test an artifact: https://github.com/adafruit/circuitpython/actions/runs/12679605089

mikeysklar, 2025-01-09:
Thx dan. Looks like you got it. I tried with your current artifact on a ESP32-S2 REV TFT. Looks good picking up the local I2C battery monitor and a MCP23008 board I plugged in.

```
Adafruit CircuitPython 9.2.1-58-g9251aee706 on 2025-01-08; Adafruit Feather ESP32-S2 Reverse TFT with ESP32S2
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Checking board.I2C()...ADDED.
Checking board.STEMMA_I2C()...ADDED.
Checking busio.I2C(board.GP1, board.GP0)...SKIPPED: 'module' object has no attribute 'GP1'
----------------------------------------
I2C SCAN
----------------------------------------
board.I2C() addresses found: ['0x20', '0x36']
board.STEMMA_I2C() addresses found: ['0x20', '0x36']
board.I2C() addresses found: ['0x20', '0x36']
```

T-Mosher, 2025-01-09:
I tried the patched build on my stacked set of Feather boards, it worked fine.

PR #9942 (2025-01-09): ports/espressif/common-hal/busio/I2C.c: add delay after probe
- Fixes #9931 

ESP32-S2 started returning bogus I2C probe successes after #9671, when I switched to using `i2c_master_probe()` instead of using our home-grown bus probe.

I inserted a debugging print and the problem went away. That suggested a timing issue. I added a 1ms delay and it's now fine.

I tested ESP32-S3, ESP32-C6, and ESP32, and they all don't have this problem. I could conditionalize the delay to just ESP32-S2, but it's a short delay and might prevent future problems.

I also wondered whether the problem was peculiar to [... cut]
closed 2025-01-09 by dhalbert