adafruit/circuitpython #9935: Fresh CircuitPython install on Pico can't edit code

byteit101, 2025-01-04:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.1 on 2024-11-20; Raspberry Pi Pico with rp2040
Adafruit CircuitPython 9.1.4 on 2024-09-17; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
print("My Hello World!")
```

### Behavior

Neither Mu nor the file manager can save this code to the pico.
I carefully followed https://learn.adafruit.com/getting-started-with-raspberry-pi-pico-circuitpython/circuitpython-programming-basics and I fail at the step `Save the file`. The drive is read-only!

Googling around, it looks like boot.py is misconfigured, except that I don't have boot.py in the CIRCUITPY drive
Attempting to use the boot.py lines in the terminal, I get:

>>> import storage
>>> storage.remount("/", readonly=True)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Cannot remount '/' when visible via USB.

Googling this leads to results of "edit your boot.py file", which the documentation doesn't explain how to do when I can't save files!

This is an extremely terrible first-time user experience because following the documentation leads to a non-working pico.

### Additional information

I've tried a few versions of circuitpython, and I've even reformatted the drive a few times with

```
>>> import storage
>>> storage.erase_filesystem()
```

Nothing has affected this issue

byteit101, 2025-01-04:
I was finally able to get a writable CircuitPY drive by using the flash_nuke.uf2 and reinstalling circuitpython.

dhalbert, 2025-01-04:
Where did your RPi Pico come from, and it is the standard one?

byteit101, 2025-01-04:
It's my usual test one, I think I originally got it at digikey?
It's had a variety of C/C++ sdk programs loaded on it before, as well as one micropython when I first got it. 

Some more details because I was very tired last night:
Once I loaded CircuitPython on it, the drive showed up with the default files. I was also trying to follow https://learn.adafruit.com/mp3-playback-rp2040/pico-i2s-mp3 at the same time. I mounted circuitpy and tried dropping the pin-detector example (and possibly the wav file, I don't recall exactly) from that page into the drive. My file manager gave me an error, and the drive unmounted. I remounted the drive, now not called circuitpy, and now with only one file (`boot_out.txt`)
A this point, trying to transfer the wav, the mp3, or any .py files lead to it stalling and erroring out after some time (something about unable to transfer file).

I opened up the REPL, and eventually found this:
```
>>> os.listdir()
['?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', '?', 'boot_out.txt']
>>> import microcontroller
>>> microcontroller.on_next_reset(microcontroller.RunMode.UF2)
>>> microcontroller.reset()
```
This was when I downgraded the first time, and loaded a new Circuitpython version. As that didn't fix the issue, eventually I found and executed:
```

closed 2026-02-08 by dhalbert