adafruit/circuitpython #9937: ESP32-S3 Safe Mode when using HTTPS / TLS TCP server

anecdata, 2025-01-04:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.1-45-ge596261dc1-dirty on 2024-12-29; Adafruit Feather ESP32-S3 TFT with ESP32S3

Adafruit CircuitPython 9.2.1-43-g2c720d9161-dirty on 2024-12-28; Waveshare ESP32-S3-ETH with ESP32S3
```

### Code/REPL

```python
Various comments and example code using `adafruit_httpserver` library and manual TLS TCP socket server at: https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/pull/88

Using latest HTTP Server library from that PR. Safe mode occurs with 9.2.1, or with `DEBUG` build from recent tip of main.

It is *not* necessary to have any client attempt to connect.
```

### Behavior

Intermittent safe mode when running HTTPS servers, can be several minutes or a couple hours. `DEBUG` output shows crashes on Core 0 or on Core 1, with varying error messages. Not a lot of consistency. `DEBUG` backtraces are either corrupted or yield no useful output ( `decode_backtrace.py` just prints a number of blank lines).

Changing to a blocking `accept()` call _seems_ to fix the issue, or at least I have not observed any Safe Mode occurrences when using blocking `accept()`.

Tested predominantly on adafruit_feather_esp32s3_tft and waveshare_esp32_s3_eth (using wifi). @michalpokusa observed the issue on a Memento board.

<details>
<summary>Example adafruit_feather_esp32s3_tft debug output</summary>

```
W (13831) CP wifi: got ip
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: BREAK instr 
Core  1 register dump:
PC      : 0x403743c0  PS      : 0x00060536  A0      : 0x4038d768  A1      : 0x3fca4020  
A2      : 0x3c1afb40  A3      : 0x3c1a15b4  A4      : 0x0000003f  A5      : 0x00000030  
A6      : 0x0000000c  A7      : 0x00000df9  A8      : 0x00000001  A9      : 0x3fcac68c  
A10     : 0x00000001  A11     : 0x3fcb7888  A12     : 0x80384cbd  A13     : 0x3fcba5f0  
A14     : 0x00060023  A15     : 0x00000003  SAR     : 0x00000019  EXCCAUSE: 0x00000001  
EXCVADDR: 0x4038d738  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x403743bd:0x3fca4020 0x4038d765:0x4038d768 |<-CORRUPTED

ELF file SHA256: 30a3826f5

CPU halted.

- - - -        

W (13763) CP wifi: got ip
Guru Meditation Error: Core  0 panic'ed (Double exception). 

Core  0 register dump:
PC      : 0x4038db12  PS      : 0x00040e36  A0      : 0x00000000  A1      : 0x3fcba710  
A2      : 0x00060523  A3      : 0x00000000  A4      : 0x00060520  A5      : 0x00000004  
A6      : 0x3fcac880  A7      : 0x3fcac884  A8      : 0x600c0034  A9      : 0x00000001  
A10     : 0x3fc9e6cc  A11     : 0xffffffff  A12     : 0xb33fffff  A13     : 0x00060523  
A14     : 0x0000001e  A15     : 0x0000abab  SAR     : 0x0000001c  EXCCAUSE: 0x00000002  
EXCVADDR: 0x00060513  LBEG    : 0x420a8e74  LEND    : 0x420a8e87  LCOUNT  : 0x3fcba7b0  

Backtrace: 0x4038db0f:0x3fcba710

ELF file SHA256: 30a3826f5

CPU halted.

- - - -

W (13840) CP wifi: got ip
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
[... 82 lines cut ...]
Core  1 register dump:
PC      : 0x400570ec  PS      : 0x00060f34  A0      : 0x82003f11  A1      : 0x3fcc0910  
A2      : 0x3c1f5d70  A3      : 0x00000000  A4      : 0x00000030  A5      : 0x3c1f5d80  
A6      : 0x00000002  A7      : 0x00000003  A8      : 0x8200ec58  A9      : 0x3fcc0a10  
A10     : 0x3c19087c  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3c1efeb8  
A14     : 0x3c1f7d70  A15     : 0x3c1f7c0d  SAR     : 0x0000001c  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000001  

Backtrace: 0x400570e9:0x3fcc0910 0x42003f0e:0x3fcc0920 0x42003339:0x3fcc0960 0x420113f1:0x3fcc0980 0x4204cded:0x3fcc09a0 0x4203ef52:0x3fcc09c0 0x42015026:0x3fcc09f0 0x4200eb8d:0x3fcc0a10 0x4200ec55:0x3fcc0a30 0x4206a8f7:0x3fcc0a50 0x4206affd:0x3fcc0a70 0x42058249:0x3fcc0aa0 0x42015026:0x3fcc0ac0 0x4200eb8d:0x3fcc0ae0 0x4200ec55:0x3fcc0b00 0x420203e9:0x3fcc0b20 0x420151d9:0x3fcc0bc0 0x4200eb8d:0x3fcc0be0 0x4200ec55:0x3fcc0c00 0x420203e9:0x3fcc0c20 0x420151d9:0x3fcc0cc0 0x4200eb8d:0x3fcc0ce0 0x4200eba2:0x3fcc0d00 0x42071c36:0x3fcc0d20 0x4207205e:0x3fcc0dc0 0x420245c1:0x3fcc0de0 0x42024ab9:0x3fcc0e00 0x42025006:0x3fcc0e60 0x42026d2a:0x3fcc0e90 0x42168480:0x3fcc0eb0

Core  0 register dump:
PC      : 0x403823ca  PS      : 0x00060634  A0      : 0x820a9fed  A1      : 0x3fcc15c0  
A2      : 0x00060623  A3      : 0x00000000  A4      : 0x00060620  A5      : 0x03ca3430  
A6      : 0x3fcbcea8  A7      : 0x3fcbce90  A8      : 0x80378eae  A9      : 0x3fcc15a0  
A10     : 0x3fcbcea8  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00060023  A15     : 0x00000003  SAR     : 0x0000001d  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x403823c7:0x3fcc15c0 0x420a9fea:0x3fcc15e0 0x420ab3af:0x3fcc1600 0x4038775b:0x3fcc1620

ELF file SHA256: cb964ad5b

CPU halted.
```
</details>

michalpokusa, 2025-01-05:
Possibly related https://github.com/adafruit/circuitpython/issues/9003

anecdata, 2025-01-06:
Tested ESP32-S2 (Feather TFT) with the fix from "espressif: Allow mbedtls data in psram if available" #9867. The memory fix allows HTTPS server to run on ESP32-S2, but S2 is prone to intermittent Safe Mode like S3.

dhalbert, 2025-01-06:
This still sounds like there is some stray pointer or use-after-free or something going on. The ability to use PSRAM may make the bug less likely.

anecdata, 2025-01-06:
In continued testing, I've still seen no safe modes with blocking `accept()`. Safe mode seems to be correlated with non-blocking `accept()`.

eightycc, 2025-03-27:
Generated an ELF map for the `waveshare_esp32_s3_eth` at the documented commit hash with `DEBUG=1`. In all 3 cases, watchdog is firing on one of the CPUs while the other is looping in a non-code area of flash. Ignoring the top backtrace frame (which is garbage), and manually reconstructing, looping CPU is going into the weeds in either `gc_alloc` or `gc_realloc` called from `socketpool_socket_recv_into()` called from `socketpool_socket_accept()`.

Likely same bug as #9003.

eightycc, 2025-03-28:
This could be a memory exhaustion issue. In `common_hal_socketpool_socket()` `mp_obj_malloc_with_finaliser()` is called to allocate a new `socketpool_socket_obj_t`, but the return is not tested for NULL.

eightycc, 2025-03-28:
With some debug tracing added and using the test server from #9428, I'm seeing `socketpool_socket_accept()` getting invoked almost continuously, each time allocating another `socketpool_socket_obj_t`. This is due to polling from `adafruit_httpserver/server.py` at line 503. Although not an error, it's definitely putting memory management through its paces. I suspect it's related. Set up now to trap an allocation failure and running...

eightycc, 2025-03-28:
Can reproduce, but did not trap on NULL allocation. I'm starting to suspect that this issue, plus #9003, #9428, #9460, and #9173 are all the same memory management bug masquerading as a networking issue.

anecdata, 2025-03-28:
#9173 is the only one that does not use httpserver / accept.

anecdata, 2025-03-28:
Occurred to me that the observation I made about blocking versus non-blocking may just be a matter of time. Non-blocking hammers on the accept call, but blocking may eventually crash too though in orders of magnitude more time.

eightycc, 2025-03-28:
> https://github.com/adafruit/circuitpython/issues/9173 is the only one that does not use httpserver / accept.
Right. #9173 could be a different issue.

eightycc, 2025-04-01:
After a fruitful chat with @dhalbert getting some traction on debug. Found multi-heap integrity testing code in IDF: `heap_caps_check_integrity_all()`. Added calls to this in `port_malloc()`, `port_free()`, and `port_realloc()`. After about 1/2 hour running simple server from #9428 I'm getting this:
```

Core  1 register dump:
PC      : 0x4039367a  PS      : 0x00040036  A0      : 0x00000000  A1      : 0x3fcaeb10  
A2      : 0xffffffff  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3fcb4b0c  
A6      : 0x00000001  A7      : 0x3fcaede8  A8      : 0x8202c16d  A9      : 0x3fcaeaf0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3fca386c  A15     : 0x3fca386c  SAR     : 0x00000010  EXCCAUSE: 0x00000002  
EXCVADDR: 0xfffffff0  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x40393677:0x3fcaeb10

ELF file SHA256: 07f337e89

CPU halted.
```
The PC is trash, but the stack pointer is consistent for the 3 failures I've produced so far. To me this suggests that `heap_caps_check_integrity_all()` is crashing due to stack corruption. I'll need to revisit JTAG to access the OCD on the Xtensa core to get further.

eightycc, 2025-04-04:
Was able to catch a failure with gdb over JTAG running. In the case I caught, crash was due to a double fault on thread 4:
```
Thread 4 (Thread 1070344812 "No Name" (Name: No Name, State: Running @CPU1)):
#0  _DoubleExceptionVector () at /home/rabeles/Development/cp-development/issue-9937/cp-repo-92x-9937/ports/espressif/esp-idf/components/xtensa/xtensa_vectors.S:564
#1  0x600c0034 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```
In this instance the thread is crashing on a memory access error due to a corrupted stack pointer:
```
(gdb) p/x $sp
$1 = 0x1
```
Working backwards, the `DEPC` register (the double-faulting instruction pointer) points to the first instruction of `WindowOverflow12` a handler for a register window overflow on a `CALL12` op. From there the trail goes cold. Register `A12` should contain the caller's PC, but it points to an invalid location.

Sleuthing the registers that do point to valid memory, `A2` points to `reason_spinlock` the contents indicating that the spinlock is free. `A9` points into `usb_device_stack`. This hints at a recently completed cross-core signal involving `tiny_usb` which is currently executing on thread 9.

The crashes I've observed do vary, so I'm running again hoping for a crash with more context.

eightycc, 2025-04-04:
Breaking into a running system, thread 4 is the CircuitPython thread. Thread 9 (the tiny_usb thread) shows a valid stack, whereas for the crash case it didn't.

eightycc, 2025-04-06:
Had a few fresh ideas. Found the CP task's stack by exploring its OpenRTOS TCB, and then decoded it manually:
```
tos:                                0x403818f4	0x4002e2a8
0x3fcc1f98:	0x00060e30	0x820032d8	0x3fcc2050	0x3fcc2178 m_alloc
0x3fcc1fa8:	0x00000001	0x00000001	0x00000004	0x3c1ee794
0x3fcc1fb8:	0x3c1efca0	0x00000000	0x3fcc2020	0x3c1efca8
0x3fcc1fc8:	0x3fcc2054	0x00000004	0x3c1efcac	0x00000017
0x3fcc1fd8:	0x00000000	0x0000001e	0x00000000	0x00000000
0x3fcc1fe8:	0x40056f5c	0x40056f72	0xffffffff	0x40381c18
0x3fcc1ff8:	0x3c1efcac	0x4038a276	0x3fc605c8	0x00000000
0x3fcc2008:	0x00000000	0x00000000	0xb33fffff	0x00000000
0x3fcc2018:	0x00000000	0x00000000	0x00000000	0x400570ec
0x3fcc2028:	0x00060430	0x82003efd	0x3fcc20e0	0x00000000 gc_alloc
0x3fcc2038:	0x00000000	0x00000010	0x82015208	0x3fcc2170 fun_bc_call
0x3fcc2048:	0x3c1e1608	0x00000000	0x820100a0	0x3fcc2080 mp_raise_OSError
0x3fcc2058:	0x3c182274	0x00000017	0x00000000	0x00000000
0x3fcc2068:	0x00000000	0x3fcc2170	0x8204d27a	0x3fcc20a0 common_hal_socketpool_socket_accept
0x3fcc2078:	0x00000016	0x02e5ca6f	0x3fcc208c	0x00000000
0x3fcc2088:	0x0000000b	0x3fcb51f8	0x8203f421	0x3fcc20c0 _socketpool_socket_accept
0x3fcc2098:	0x3c1e4820	0x3fcc20e4	0x00000000	0x00000000
0x3fcc20a8:	0x00000000	0x3c1e4820	0x82015055	0x3fcc20e0 fun_builtin_1_call
```

The `mp_raise_OSError` is a debug call I added to catch a failure of `mp_obj_malloc_with_finaliser` to allocate a new `socketpool_socket_obj_t`. The failure occurs while subsequently attempting to raise an `ENOMEM OSerror`.

Manually examining the code, I think it is plausible that a race condition exists between `gc_alloc` and `mp_irq_handler` over `gc_lock_depth` that is causing spurious NULL returns from `gc_alloc`. I've set up additional trapping to see if this is the case.

eightycc, 2025-05-07:
ESP-IDF component `esp_rom` contains a patch to `longjmp` for Xtensa that creates a critical section around modification of the register window's start register. CircuitPython does not incorporate this patch.

Comments from the code in `esp-idf/components/esp_rom/patches/esp_rom_longjmp.S`:

"This file contains a modified version of the original Xtensa longjmp implementation. In this modified version, setting WINDOWSTART = 1 << WINDOWBASE is done inside a critical section.
This is necessary because after a FreeRTOS context switch in IDF, the values of WINDOWBASE and WINDOWSTART are not guaranteed to be the same as before the context switch."

Because CircuitPython disables native NLR (Non-Local Return) for Espressif Xtensa parts such as the ESP32-S3, it makes heavy use of Newlib's un-patched `setjmp/longjmp` resident in the ESP's ROM. This exposes CircuitPython to register window corruption when an interrupt occurs while running in the un-patched `longjmp` function.

Many of the observed failures involve a double-fault during register window manipulation.

eightycc, 2025-05-08:
Retested with CP 10.0.0-alpha.4, double fault after about 2 hours. Problem still there. ESP-IDF 5.4.1 did not resolve it. I didn't expect it to, but it was worth a try.

Moving on to test with patched `longjmp`.

eightycc, 2025-05-08:
After 14 hours, no failure with the patched `longjmp`. I'm going to let it run for another 24 hours to be certain, then push the PR.

anecdata, 2025-05-08:
This sounds very promising. Can you think of an obvious reason this occurs only with HTTPS, or could it just be a fluke that we haven't observed it with HTTP yet? I'll set up to test a few variations this weekend.

eightycc, 2025-05-08:
> Can you think of an obvious reason this occurs only with HTTPS, or could it just be a fluke that we haven't observed it with HTTP yet?

This happens with HTTP, too. The test case I've been working with is a simple HTTP server from #9428. The root cause isn't a networking issue at all, which is kind of surprising. Instead, it's due to the socket polling loop using NLR (Non-Local Return) for every iteration that eventually corrupts the Xtensa windowing registers. In its implementation, NLR uses `longjmp`. When an interrupt happens that triggers a context switch in FreeRTOS at just the right spot in `longjmp`, it corrupts the windowing registers. Espressif knows this, and so they provide a wrapper for `longjmp`. Because CP uses its own build tools, it never picked up wrapper. I only found it because I was looking into the ESP-IDF ROM component for another issue, and came across the wrapper.

eightycc, 2025-05-09:
@anecdata It's been more than 24 hours, and the test is still running without failure. If it's still running tomorrow morning, I'll push the PR. Besides this issue, this will fix #9428. It is also likely to fix #9003 and #9460. There may be others as well.

eightycc, 2025-05-09:
@anecdata The test is still going strong after 36+ hours. The build artifacts are in PR #10328.

anecdata, 2025-05-09:
@eightycc Sounds great! I'm tied up much of today, but will run it later.

closed 2025-05-09 by tannewt