adafruit/circuitpython #9952: Serial console broken on 9.2.2 on T-Display-S3

rmrfus, 2025-01-12:
### CircuitPython version

```python
Unable to access, but used adafruit-circuitpython-lilygo_tdisplay_s3-en_US-9.2.2.bin image.
```

### Code/REPL

```python
n/a
```

### Behavior

Neither serial console nor ampy doesn't work on image adafruit-circuitpython-lilygo_tdisplay_s3-en_US-9.2.2.bin - repl is displayed on the screen, but serial on 115200 doesn't work.

Rollback to adafruit-circuitpython-lilygo_tdisplay_s3-en_US-9.2.1.bin image fix the problem.

Was able to consistently repro the problem by switching between 9.2.2 and 9.2.1 images.

dhalbert, 2025-01-13:
This is probably a duplicate of #9956.

dhalbert, 2025-01-14:
Closing as duplicate.

closed 2025-01-14 by dhalbert