adafruit/circuitpython #9953: monotonic_ns jumps

dheera, 2025-01-12:
### CircuitPython version

```python
6.0.0
```

### Code/REPL

```python
import time
for i in range(1000000000):
    print(time.monotonic_ns())
    time.sleep(0.001)
```

### Behavior

```
1431848724366
1431849700929
1431850677491
1431851654054
1431852630616
1431853607179
1431854583741
1431855560304
1431856536866
1431857513429
1431858520509
1431859466554
1432001098634 <-- wtf?
1432002044679
1432003021241
1432003997804
1432004974366
1432005981447
1432006927491
1432007904054
1432008911134
1432009857179
1432010833741
1432011810304
1432012786866

```

### Description

It seems that the nanoseconds part is counting slower than it should be

dhalbert, 2025-01-12:
CircuitPython is not a real-time system. Some interrupt or garbage-collection could have occurred that caused a delay. It could also be an old bug that was fixed. You said you are using 6.0.0, which is from 2020. Also please report what board you are using.

dheera, 2025-01-12:
Hi,
This is a Feather S2.
It appears to happen every second, the ~850000 to ~999999 are missing. It's extremely regular and there isn't a noticeable execution delay so doesn't appear to be a garbage collection phenomenon.

dhalbert, 2025-01-12:
Could you try 9.2.2 instead of 6.0.0?

dhalbert, 2025-07-29:
@dheera Did you try a newer version of CircuitPython? At this point, I would recommend trying 10.0.0-beta.0 or later as well.