adafruit/circuitpython #9954: Mount point directory is not hidden when a filesystem is mounted on it

dhalbert, 2025-01-12:
Noted as a result of https://forums.adafruit.com/viewtopic.php?t=215933

```py
Adafruit CircuitPython 9.2.1-54-g26283dcb02 on 2025-01-06; Adafruit Metro ESP32S3 with ESP32S3
>>> import sdcardio,storage,board,os
>>> os.listdir()
['.fseventsd', '.metadata_never_index', '.Trashes', '.Trash-1000', 'sd', 'settings.toml', 'lib', 'boot_out.txt',  'ble_uart_echo_client.py']
>>> sdcard = sdcardio.SDCard(board.SPI(), board.SD_CS)
>>> vfs = storage.VfsFat(sdcard)
>>> storage.mount(vfs,"/sd")
>>> os.listdir()
# Now there are two 'sd` entries.
['sd', '.fseventsd', '.metadata_never_index', '.Trashes', '.Trash-1000', 'sd', 'settings.toml', 'lib', 'boot_out.txt',  'ble_uart_echo_client.py']
# But in this filesystem, there is only one.
>>> os.listdir(".")
['.fseventsd', '.metadata_never_index', '.Trashes', '.Trash-1000', 'sd', 'settings.toml', 'lib', 'boot_out.txt',  'ble_uart_echo_client.py']
>>> 
```

RetiredWizard, 2025-01-12:
This might be simpler to solve than the other VfsFat mount point issues, but just in case someone's digging around, I figured I'd link a couple possibly related issues #9045 and #8409

dhalbert, 2025-01-12:
Though the extra listing is confusing, it does not seem to affect actual functionality.

bablokb, 2025-03-12:
This problem should be solved, because of the recent PR from @tannewt (#10122). Now we have a saves-partition which is internally mounted on `/saves`. If you mount your CIRCUITPY-drive from the host computer, you can see the files within
 `/saves`, you can add and delete and modify files, although the C-source code promises something different: https://github.com/adafruit/circuitpython/blob/ca0eec3f1a3dec49245a9fd5883e40eb9d96e2a8/supervisor/shared/filesystem.c#L96.

But every change within `/saves` only changes the FAT on the the CIRCUITPY-drive, not on the internal mounted second partition, so the promise is true. It *is* creating two different truths about the `/saves` directory though and this will certainly confuse users.

Basically it just does not work to mount the same storage from two computers without causing problems. That is why we have network and cluster filesystems, that take care of keeping the truth.

The clean solution would probably to expose the second partition as a read-only partition to the host and let the host do the mounting. A workaround would be to change all the filesystem-related APIs, i.e. not only read the FAT, but also filter all entries that are hidden by mounted filesystems. This won't change the user-view, but the view from within CP will at least be correct.

jepler, 2025-03-12:
I didn't check but I suspect the duplicate directory listing in `os.listdir()` is coming from this code block in `extmod/vfs.c`:
```
389         } else {
390             // a mounted directory
391             mp_obj_tuple_t *t = MP_OBJ_TO_PTR(mp_obj_new_tuple(3, NULL));
```
since we've decided mount points should exist as directories in CIRCUITPY, we might just want to remove that block as a CIRCUITPY-CHANGE.

jepler, 2025-03-12:
@bablokb This PR is about how the filesystem shows up in CircuitPython using functions like `os.listdir`. It's not about how the filesystem shows up on a host computer when mounted as a drive. The fact that the host computer will see the placeholder file in `saves` on the CIRCUITPY drive, instead of the real save content, would be a separate issue (and likely one without a good easy solution, sadly; but that may just be my natural pessimism showing)

tannewt, 2025-03-12:
> since we've decided mount points should exist as directories in CIRCUITPY, we might just want to remove that block as a CIRCUITPY-CHANGE.

I'll look into this. I'm looking at SD card automounting again.

PR #10129 (2025-04-02): Automount SD cards
And make them available over USB MSC. They can be remount to make them writable (slowly) from the host PC.

Fixes #9954. Fixes #8678. Fixes #3477
commit d0c84009 (2025-04-02): Automount SD cards

And make them available over USB MSC. They can be remount to make
them writable (slowly) from the host PC.

Fixes #9954. Fixes #8678. Fixes #3477
closed 2025-04-02 by dhalbert